281677160/build-actions #2: 请教大佬一个问题

drw2005, 2021-08-18:
[image]
本地编译，想把上面红圈处改回默认名字，如何操作？

改密码还改回管理权
终端改回TTYD终端

感谢！

281677160, 2021-08-18:
本地编译改不了，你要改就用云编译吧

drw2005, 2021-08-18:
> 本地编译改不了，你要改就用云编译吧

知道了。我现在云编译也用不了，被限制了，没办法，只能本地。

drw2005, 2021-08-19:
[image]

另外，再提个小问题，LEDE编译的上图中有个小尾巴。但是immortalwrt编译的就没有。
只是个小问题，大佬闲暇时帮看看。

281677160, 2021-08-19:
你这小问题，我是没办法解决了，我早就说过了，老早之前就已经设置好了，我编译出来的固件也不会这样
给你几个删除命令，删除了全部编译文件，你再用第三个编译命令测试吧，还不行我也是没办法的

rm -rf {dl,.bf_config,compile.sh,.compile}
rm -rf {openwrt,.Lede_core,.Lienol_core,.amlogic_core,.Mortal_core}
bash <(curl -fsSL git.io/JcGDV)

drw2005, 2021-08-20:
> 你这小问题，我是没办法解决了，我早就说过了，老早之前就已经设置好了，我编译出来的固件也不会这样
> 给你几个删除命令，删除了全部编译文件，你再用第三个编译命令测试吧，还不行我也是没办法的
> 
> rm -rf {dl,.bf_config,compile.sh,.compile}
> rm -rf {openwrt,.Lede_core,.Lienol_core,.amlogic_core,.Mortal_core}
> bash <(curl -fsSL git.io/JcGDV)

这两个命令管用，小尾巴没有了，已经很完美了，谢谢大佬！

281677160, 2021-08-20:
可以就行了，不行我还真没办法，不用客气了

drw2005, 2021-08-30:
又有个问题请教：我平时编译用的都是LEDE源码，后来发现4月份编译的显示有端口状态和IPV6信息，如图一红框处。然后最近编译的这两项都没有了，如图二。请教大佬这是咋回事？是我编译时哪里设置的不对？谢谢！
[image]

[image]

281677160, 2021-08-30:
你直接物理安装，然后拨号就应该啥都有了

drw2005, 2021-08-31:
手头有个阿里云服务器，上海线路，装有ubuntu20.04，但是没办法科学上网（主要是不知道在ubuntu下如何配置科学上网），执行不了bash <(curl -fsSL git.io/JcGDV)，请问大佬有没有啥办法？

281677160, 2021-08-31:
没法科学上网搞不了

drw2005, 2021-09-07:
请问大佬，DDNSTO插件取消了？怎么找不到了？这个还比较常用，虽然可以自己单独安装，但是还是希望能集成进去。
谢谢！

281677160, 2021-09-07:
应该加好了