281677160/build-actions #13: 请教如何改SSH连接

gdstchdr1, 2021-09-30:
如题，可以配置一下自己要的插件，谢谢！！

281677160, 2021-09-30:
没看明白你说的是啥意思？要开启SSH链接？
说明都写了很清楚了，开启SSH连接有两个地方都可以开启，一个是启动时候随便开启 https://github.com/danshui-git/shuoming/blob/master/%E6%89%8B%E5%8A%A8%E5%BC%80%E5%A7%8B.md
一个是 build/对应源码/settings.ini文件

gdstchdr1, 2021-09-30:
谢谢大佬解答，是我没表达清楚，是要开启SSH链接，已经解决了，谢谢。

Myoko, 2021-10-15:
听说连接SSH 会封号！还是建议不要连接的好

281677160, 2021-10-16:
> 听说连接SSH 会封号！还是建议不要连接的好

现在一般不封号，SSH后直接编译会给停止编译就是，进SSH弄好配置文件，关闭编译，再把配置文件覆盖到你需要编译的对应源码的.config里面再编译就好了

281677160, 2021-10-16:
> 听说连接SSH 会封号！还是建议不要连接的好

要看我写的说明，你这个就是压根不明白人家为什么进SSH的原因，所以才这样说

ywyha, 2021-10-17:
> > 听说连接SSH 会封号！还是建议不要连接的好
> 
> 要看我写的说明，你这个就是压根不明白人家为什么进SSH的原因，所以才这样说

大佬，怎样修改链接SSH等待时间，默认好像是等待半个小时……，可不可以让他跳过或者缩短这步骤……

ywyha, 2021-10-17:
> > > > 听说连接SSH 会封号！还是建议不要连接的好
> > > 
> > > 
> > > 要看我写的说明，你这个就是压根不明白人家为什么进SSH的原因，所以才这样说
> > 
> > 
> > 大佬，怎样修改链接SSH等待时间，默认好像是等待半个小时……，可不可以让他跳过或者缩短这步骤……
> 
> 你意思是你现在打开了SSH连接配置固件，但是不懂怎么关了它了？把true改回false不就关闭了？为什么要等待？还是你进了SSH配置固件后，不懂结束？看说明啊 https://github.com/danshui-git/shuoming/blob/master/3SSH%E8%BF%9E%E6%8E%A5%E8%AF%B4%E6%98%8E.md

谢谢大佬，明白了，为我等小白操碎了心，感谢提供这么好用的工具，谢谢。

Myoko, 2021-10-18:
> > 听说连接SSH 会封号！还是建议不要连接的好
> 
> 要看我写的说明，你这个就是压根不明白人家为什么进SSH的原因，所以才这样说

抱歉！我是看到有几个人说他们进SSH 后被封了 所以我才觉得这个操作有风险 感谢解答

ywyha, 2021-10-20:
> > > > > > 听说连接SSH 会封号！还是建议不要连接的好
> > > > > 
> > > > > 
> > > > > 要看我写的说明，你这个就是压根不明白人家为什么进SSH的原因，所以才这样说
> > > > 
> > > > 
> > > > 大佬，怎样修改链接SSH等待时间，默认好像是等待半个小时……，可不可以让他跳过或者缩短这步骤……
> > > 
> > > 
> > > 你意思是你现在打开了SSH连接配置固件，但是不懂怎么关了它了？把true改回false不就关闭了？为什么要等待？还是你进了SSH配置固件后，不懂结束？看说明啊 https://github.com/danshui-git/shuoming/blob/master/3SSH%E8%BF%9E%E6%8E%A5%E8%AF%B4%E6%98%8E.md
> > 
> > 
> > 谢谢大佬，明白了，为我等小白操碎了心，感谢提供这么好用的工具，谢谢。
> 
> 不客气，我正头痛呢，电报群给个傻逼举报了，唉 https://github.com/281677160/openwrt-package/issues 在这里提了好几个issues呢，这样的大爷真伤不起

我从去年起接触这个脚本后，问了大佬数次问题，而且都是小白问题，每次都是有问必答，现在30多岁，需要平衡生活时间和互联网时间，所以没法像头些年那样有足够的时间和精力去了解互联网的一些新东西，这种大佬这种脚本出现真是解决了非常大的困扰，还想用新东西，还不想加入自己不想要的东西，然后还不懂技术，然后还没时间和精力去了解和学习这些东西，大佬这个脚本完全解决了自己的需求，真的很感谢。今年群里的情况我也看到了，去年就是被这种人弄的删库“跑路”一路还难过一段时间(ಥ﹏ಥ)，确实败坏心情。

eightsheep67, 2021-12-11:
> > 听说连接SSH 会封号！还是建议不要连接的好
> 
> 现在一般不封号，SSH后直接编译会给停止编译就是，进SSH弄好配置文件，关闭编译，再把配置文件覆盖到你需要编译的对应源码的.config里面再编译就好了

我通过网页进入SSH弄完配置文件，如何导出.config文件

281677160, 2021-12-11:
> > > 听说连接SSH 会封号！还是建议不要连接的好
> > 
> > 
> > 现在一般不封号，SSH后直接编译会给停止编译就是，进SSH弄好配置文件，关闭编译，再把配置文件覆盖到你需要编译的对应源码的.config里面再编译就好了
> 
> 通过网页进入SSH弄完配置文件，如何导出.config文件 最近真的SSH过的flow都会被掐掉，源码的默认配置又改了，我想稍微更改一下配置好难……

看教程，SSH跟编译是分开的

eightsheep67, 2021-12-11:
> > 听说连接SSH 会封号！还是建议不要连接的好
> 
> 现在一般不封号，SSH后直接编译会给停止编译就是，进SSH弄好配置文件，关闭编译，再把配置文件覆盖到你需要编译的对应源码的.config里面再编译就好了

想问的就是这个步骤有点不太明白
1.关闭编译意思是手动cancel掉这个workflow吗？这样还会生成配置文件吗？
2.必须用你仓库的代码才可以吗，还是所有的Github Action云编译代码都可以，我的是用P大的代码

281677160, 2021-12-11:
脚本老早就改了，如果你有认真看说明，就不会问出这样的问题

281677160, 2021-12-11:
[image]

eightsheep67, 2021-12-11:
原来是这样，那我也研究更新下

closed 2022-01-11 by 281677160