281677160/build-actions #19: 报错这个是啥原因大佬

rq1025330, 2021-10-11:
[image]

rq1025330, 2021-10-11:
哦哦  自从前段时间开始都是显示超时 以前不得

281677160, 2021-10-11:
或者你把日志下载下来看看，有没有其他的错误导致写日志超时，怎么下载日志，我有说明

rq1025330, 2021-10-11:
好

rq1025330, 2021-10-11:
大佬回复真快。记起来了之前还用过你的插件库，后来我又全部改成单独插件拉取了，大佬重复的就不能再拉取了吗，是重新拉取会覆盖，还是那个有优先权？能改成lede原始的吗?，然后要啥添加啥。
[image]
h还有这里这样修改会有问题？

281677160, 2021-10-18:
你要追新的，就编译出tar.gz文件自己打包了，我这个也是拉别人的，别人整理，看内核版本，在我说明那里就有个链接进去看了
https://github.com/danshui-git/shuoming/blob/master/Amlogic.md

 
有什么方法可以快速看到你编译内核版本呢，我是本地看到文件的

rq1025330, 2021-10-18:
[image]
mei'kan没看到呀

281677160, 2021-10-18:
我不知道你要啥，呃，你自己慢慢研究吧

 
就是今天去没看到，本地没看到配置的，我一个一个看的

rq1025330, 2021-10-18:
就是吧那个插件添加到云编译里面，这样我就可以在配置里选中就可以编译了出来了，就相当于自己添加里面没有自己有需要的插件。
[image]

281677160, 2021-10-31:
你的报错，我找两天，发现是kodexplorer跟你现在不知道选的什么东西冲突了，我整理了一下你的配置文件，就可以编译了，但是也不能直接编译，直接编译超时，你要先编译一次我默认的配置文件，再来编译你新的文件，要先弄一个缓存，有缓存才能加速，我以前就跟你说过了，你没当回事

281677160, 2021-10-31:
```sh
CONFIG_TARGET_armvirt=y
CONFIG_TARGET_armvirt_64=y
CONFIG_TARGET_armvirt_64_Default=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_NOXML=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_WEBSOCKET=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_HYSTERIA_COMPRESS_UPX=y
CONFIG_LIBMBEDTLS_HAVE_ARMV8CE_AES=y
CONFIG_NFS_KERNEL_SERVER_V4=y
CONFIG_NGINX_HEADERS_MORE=y
CONFIG_NGINX_HTTP_ACCESS=y
CONFIG_NGINX_HTTP_AUTH_BASIC=y
CONFIG_NGINX_HTTP_AUTOINDEX=y
CONFIG_NGINX_HTTP_BROWSER=y
CONFIG_NGINX_HTTP_CACHE=y
CONFIG_NGINX_HTTP_CHARSET=y
CONFIG_NGINX_HTTP_EMPTY_GIF=y
CONFIG_NGINX_HTTP_FASTCGI=y
CONFIG_NGINX_HTTP_GEO=y
CONFIG_NGINX_HTTP_GZIP=y
CONFIG_NGINX_HTTP_LIMIT_CONN=y
CONFIG_NGINX_HTTP_LIMIT_REQ=y
CONFIG_NGINX_HTTP_MAP=y
CONFIG_NGINX_HTTP_MEMCACHED=y
CONFIG_NGINX_HTTP_PROXY=y
CONFIG_NGINX_HTTP_REFERER=y
CONFIG_NGINX_HTTP_REWRITE=y
CONFIG_NGINX_HTTP_SCGI=y
CONFIG_NGINX_HTTP_SPLIT_CLIENTS=y
CONFIG_NGINX_HTTP_SSI=y
CONFIG_NGINX_HTTP_UPSTREAM_HASH=y
CONFIG_NGINX_HTTP_UPSTREAM_IP_HASH=y
CONFIG_NGINX_HTTP_UPSTREAM_KEEPALIVE=y
CONFIG_NGINX_HTTP_UPSTREAM_LEAST_CONN=y
CONFIG_NGINX_HTTP_USERID=y
CONFIG_NGINX_HTTP_UWSGI=y
CONFIG_NGINX_HTTP_V2=y
CONFIG_NGINX_NAXSI=y
CONFIG_NGINX_PCRE=y
CONFIG_NGINX_UBUS=y
CONFIG_OCSERV_HTTP_PARSER=y
CONFIG_OCSERV_PROTOBUF=y
CONFIG_OPENCONNECT_GNUTLS=y
CONFIG_OPENVPN_openssl_ENABLE_DEF_AUTH=y
CONFIG_OPENVPN_openssl_ENABLE_FRAGMENT=y
CONFIG_OPENVPN_openssl_ENABLE_LZ4=y
CONFIG_OPENVPN_openssl_ENABLE_MULTIHOME=y
CONFIG_OPENVPN_openssl_ENABLE_PF=y
CONFIG_OPENVPN_openssl_ENABLE_PORT_SHARE=y
CONFIG_OPENVPN_openssl_ENABLE_SMALL=y
CONFIG_PACKAGE_adbyby=y
CONFIG_PACKAGE_aliyundrive-webdav=y
CONFIG_PACKAGE_alsa-lib=y
CONFIG_PACKAGE_alsa-ucm-conf=y
CONFIG_PACKAGE_alsa-utils=y
CONFIG_PACKAGE_amule=y
CONFIG_PACKAGE_appfilter=y
CONFIG_PACKAGE_aria2=y
CONFIG_PACKAGE_ariang=y
CONFIG_PACKAGE_attr=y
CONFIG_PACKAGE_avahi-dbus-daemon=y
CONFIG_PACKAGE_bash=y
[... 518 lines cut ...]
CONFIG_PHP7_SYSTEMTZDATA=y
CONFIG_RPCBIND_LIBWRAP=y
CONFIG_RPCBIND_RMTCALLS=y
CONFIG_SAMBA4_SERVER_AVAHI=y
CONFIG_SAMBA4_SERVER_NETBIOS=y
CONFIG_SAMBA4_SERVER_VFS=y
CONFIG_SAMBA4_SERVER_WSDD2=y
CONFIG_SQLITE3_DYNAMIC_EXTENSIONS=y
CONFIG_SQLITE3_FTS3=y
CONFIG_SQLITE3_FTS4=y
CONFIG_SQLITE3_FTS5=y
CONFIG_SQLITE3_JSON1=y
CONFIG_SQLITE3_LIBEDIT=y
CONFIG_SQLITE3_RTREE=y
CONFIG_STRONGSWAN_ROUTING_TABLE="220"
CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO="220"
CONFIG_TROJAN_GO_COMPRESS_UPX=y
CONFIG_ZSTD_OPTIMIZE_O3=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
CONFIG_PACKAGE_UnblockNeteaseMusic=y
CONFIG_PACKAGE_UnblockNeteaseMusic-Go=y
```

281677160, 2021-10-31:
这个是我用你的配置文件弄的，加了缓存才能编译成功

rq1025330, 2021-10-31:
感谢 大佬 这两天忙都还没看呢 kod 还冲突ma

rq1025330, 2021-11-02:
[image]
没有开启结果上传了

281677160, 2021-11-02:
哥们，这个是发布固件自带，要了解

281677160, 2021-11-02:
你不开发布固件，他就不出来了

closed 2022-01-11 by 281677160