281677160/build-actions #21: 所选的插件不在编译范围内?怎么解决?谢谢!

ghaner, 2021-10-13:
[image]
[image]

281677160, 2021-10-13:
这个好解决啊，编译的时候连接SSH进去选择好插件，弄出配置文件来，然后覆盖到对应源码里面再进行编译不就行了？

主题: [281677160/build-actions] 所选的插件不在编译范围内?怎么解决?谢谢! (#21)

281677160, 2021-10-13:
请确认正确配置，谢谢，我到现在都还是老老实实的自己进SSH修改配置的，我没技术自己写
[image]
[image]

closed 2021-10-16 by 281677160