281677160/build-actions #26: 请教老大一个小米mini固件编译的问题

drw2005, 2021-11-02:
用的Lede_source源码，编译出来大概15m，刷新完固件，配置好上网环境，一切正常。但是只要一重启，所有配置就丢失了，这时查看软件包空闲空间，居然为0了，这到底是啥原因？

插件就添加了ttyd、ddnsto、VSSR和turboacc这四个。我看有些人发布的MINI固件添加十几个插件的都有，还有很多主题，也没这问题啊！

请教老大，这个是啥原因？有没有在添加上面几个插件的情况下减少固件体积的办法？

感谢！

281677160, 2021-11-02:
不了解，没这路由，也没听人说过这样情况

主题: [281677160/build-actions] 请教老大一个小米mini固件编译的问题 (Issue #26)

281677160, 2021-11-02:
百度了一下，恩山有类似的帖子，你自己看看
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=5615078&extra=page%3D1%26filter%3Dtypeid%26typeid%3D44

drw2005, 2021-11-03:
> 百度了一下，恩山有类似的帖子，你自己看看 https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=5615078&extra=page%3D1%26filter%3Dtypeid%26typeid%3D44

恩山相关帖子翻遍了，也没找到解决办法。

我之前7月份编译的都没这个问题，当时插件选的比现在的多。

281677160, 2021-11-04:
> > 百度了一下，恩山有类似的帖子，你自己看看 https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=5615078&extra=page%3D1%26filter%3Dtypeid%26typeid%3D44
> 
> 恩山相关帖子翻遍了，也没找到解决办法。
> 
> 我之前7月份编译的都没这个问题，当时插件选的比现在的多。

那你用4.14的源码编译测试一下呗

drw2005, 2021-11-05:
> > > 百度了一下，恩山有类似的帖子，你自己看看 https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=5615078&extra=page%3D1%26filter%3Dtypeid%26typeid%3D44
> > 
> > 
> > 恩山相关帖子翻遍了，也没找到解决办法。
> > 我之前7月份编译的都没这个问题，当时插件选的比现在的多。
> 
> 那你用4.14的源码编译测试一下呗

我是本地编译，请教怎么退回到4.14版本？

closed 2021-12-13 by drw2005