281677160/build-actions #34: OP_DIY文件跟openwrt文件是同一个目录里面，好像没找到啊，请教一下！

drw2005, 2021-12-11:
请教老大，本地Ubuntu一键编译脚本说明里：
问：“请在/OP_DIY/xxx_source里面设置好自定义文件”是什么意思？
答：OP_DIY文件跟openwrt文件是同一个目录里面。

OpenWrt下好像没找到OP_DIY这个目录呢？

281677160, 2021-12-11:
为什么去openwrt文件夹下找？跟它同一个目录，不是在openwrt下好不好

主题: [281677160/build-actions] OP_DIY文件跟openwrt文件是同一个目录里面，好像没找到啊，请教一下！ (Issue #34)

drw2005, 2021-12-12:
也没有：

[image]

281677160, 2021-12-12:
[image]
那我也是没办法了，子系统，服务器版本的我都测试了，没问题的，就是下载一个文件而已，你的下载不了我也不知道什么情况

281677160, 2021-12-12:
[image]

drw2005, 2021-12-13:
是我的原因，我用还是之前的一键编译命令，才看到新的bash <(curl -fsSL git.io/local.sh)
这下本地更方便了，越来越好用，谢谢老大！

closed 2021-12-13 by drw2005

drw2005, 2021-12-13:
本地要是再支持定制banner就好了，老大能实现不？