281677160/build-actions #43: 本地编译有两个问题请教大佬

drw2005, 2022-03-21:
1、最近几天，编译过两次，顺利结束后，发现OP_DIY目录下多了个.config，之前自定义文件不是config吗？不带点的。这两个现在到底是哪个在起作用啊？
2、每次都会自动把luci-app-argon-config这个编译进去，即使不选择这个插件，编译完成也有这个插件。全部清空缓存重新再来也不行。神奇！

谢谢!

drw2005, 2022-03-21:
今天编译几次都失败了，尝试清空缓存，默认编译也是失败，提示找不到：/home/ drw/ openwrt/bin/ targets/ x86/64' : No such file or directory，源码下载完，开始编译后，登陆查看了一下，确实没有bin这个文件夹。前天编译还没有这个问题。

[image]

drw2005, 2022-03-22:
另外，显示界面有重复，虽然无关紧要：
[image]

[image]

281677160, 2022-03-28:
本地编译文件全部更新过，要重新拉取全新文件使用，显示问题，应该是你的问题吧，我没试过这样的

closed 2022-03-28 by 281677160