281677160/build-actions #44: 连续两天编译失败，麻烦请看一下是什么原因

cyj265, 2022-03-30:
make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/Lede_source/openwrt/include/toplevel.mk:230: world] Error 1
make[1]: Entering directory '/Lede_source/openwrt'
make[2]: Entering directory '/Lede_source/openwrt'
make[3]: Entering directory '/Lede_source/openwrt/target/linux'
make[4]: Entering directory '/Lede_source/openwrt/target/linux/ramips'
rm -f /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/localversion
/Lede_source/openwrt/scripts/kconfig.pl  + /Lede_source/openwrt/target/linux/generic/config-5.4 /Lede_source/openwrt/target/linux/ramips/mt7621/config-5.4 > /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /Lede_source/openwrt/.config >> /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.target
/Lede_source/openwrt/scripts/package-metadata.pl kconfig /Lede_source/openwrt/tmp/.packageinfo /Lede_source/openwrt/.config 5.4 > /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.override
/Lede_source/openwrt/scripts/kconfig.pl 'm+' '+' /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.target /dev/null /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.override > /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set
mv /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.old
grep -v INITRAMFS /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.old > /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set
rm -rf /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/modules
cmp -s /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.prev || { cp /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config; cp /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.prev; }
export MAKEFLAGS= ; [ -d /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/user_headers ] || make -C /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187 KCFLAGS="-fmacro-prefix-map=/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/Lede_source/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Mar 30 10:22:37 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/Lede_source/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/Lede_source/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.187 CC="mipsel-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/user_headers headers_install
grep '=[ym]' /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.config.set | LC_ALL=C sort | /Lede_source/openwrt/staging_dir/host/bin/mkhash md5 > /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.vermagic
touch /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/.configured
rm -f /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/vmlinux /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187/System.map
make -C /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187 KCFLAGS="-fmacro-prefix-map=/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/Lede_source/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Mar 30 10:22:37 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/Lede_source/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/Lede_source/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.187 CC="mipsel-openwrt-linux-musl-gcc" olddefconfig
make[5]: Entering directory '/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187'
scripts/kconfig/conf  --olddefconfig Kconfig
net/sched/Kconfig:45: warning: menuconfig statement without prompt
net/Kconfig:478:warning: multi-line strings not supported
#
# No change to .config
#
make[5]: Leaving directory '/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187'
make -C /Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187 KCFLAGS="-fmacro-prefix-map=/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/Lede_source/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Mar 30 10:22:37 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/Lede_source/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/Lede_source/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.187 CC="mipsel-openwrt-linux-musl-gcc" vmlinux vmlinuz modules
make[5]: Entering directory '/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  CC      drivers/net/ethernet/ralink/mtk_eth_soc.o
drivers/net/ethernet/ralink/mtk_eth_soc.c: In function 'fe_init':
drivers/net/ethernet/ralink/mtk_eth_soc.c:[140](https://github.com/cyj265/build-actions/runs/5752307695?check_suite_focus=true#step:14:140)5:13: error: too few arguments to function 'of_get_mac_address'
  mac_addr = of_get_mac_address(priv->dev->of_node);
             ^~~~~~~~~~~~~~~~~~
In file included from drivers/net/ethernet/ralink/mtk_eth_soc.c:26:
./include/linux/of_net.h:14:12: note: declared here
 extern int of_get_mac_address(struct device_node *np, u8 *mac);
            ^~~~~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:262: drivers/net/ethernet/ralink/mtk_eth_soc.o] Error 1
make[8]: *** [scripts/Makefile.build:497: drivers/net/ethernet/ralink] Error 2
make[7]: *** [scripts/Makefile.build:497: drivers/net/ethernet] Error 2
make[6]: *** [scripts/Makefile.build:497: drivers/net] Error 2
make[5]: Leaving directory '/Lede_source/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.187'
[... cut ...]
time: target/linux/compile#5.39#1.81#6.84
make[3]: *** [Makefile:11: compile] Error 2
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/Lede_source/openwrt'
make[1]: *** [target/Makefile:18: /Lede_source/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/Lede_source/openwrt'
make: *** [/Lede_source/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

281677160, 2022-03-30:
ERROR: target/linux failed to build
应该是上游源码问题吧，你换源码编译试试

cyj265, 2022-03-30:
> 

好的，我试试

281677160, 2022-03-30:
用4.19的那个源码看看

closed 2022-03-30 by cyj265