281677160/build-actions #54: 云编译出现错误

wqzxhpx1988, 2022-04-21:
[image]

281677160, 2022-04-21:
同步最新仓库再试试

wqzxhpx1988, 2022-04-21:
昨天完全同步的最新仓库，从昨天开始一直出现这个错误好像是容量不够
[image]

281677160, 2022-04-21:
今天早上改了

wqzxhpx1988, 2022-04-21:
[image]

wqzxhpx1988, 2022-04-21:
我重新再同步试试

281677160, 2022-04-21:
同步完测试吧，同步后应该没问题了，昨天跟今天早上修改了几个BUG,我懒的去写公告了

wqzxhpx1988, 2022-04-21:
好的大神，我同步后再试试，谢谢

281677160, 2022-04-21:
以前每次启动都同步compile.yml文件的时候省心多了，没修改diy-part.sh文件基本每次都是最新仓库

wqzxhpx1988, 2022-04-21:
确实以前太省心了，现在还在编译，一会看看结果

281677160, 2022-04-21:
没办法，众口难调，有些人喜欢自己修改一下yml文件，显示些个性，刚修改了源码获取方式了，我都放common文件了，及时更新仓库

wqzxhpx1988, 2022-04-21:
好的 ，固件稳定比什么都强，等一会编译完看下后再更新仓库

wqzxhpx1988, 2022-04-21:
已经编译成功了，非常感谢

closed 2022-04-21 by wqzxhpx1988