281677160/build-actions #55: 有个疑惑？为什么 k2p 不管如何删减插件，固件大小总是超过 16MB?

shelken, 2022-04-23:
我用的是 Lede源码。不管如何删减插件，只保留我必须的几个插件，都还是超过限制。我在[其他仓库](https://github.com/MrH723/Actions-OpenWrt)看到一个编译 k2p 的。他的插件比我选择的多，但是却只有 13MB。内核也是 5.4 。用的也是 Lede源码的 master 分支。是不是我哪里有问题呢？这是我的 config 文件：
`CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_phicomm_k2p=y
CONFIG_PACKAGE_6in4=y
CONFIG_PACKAGE_6rd=y
CONFIG_PACKAGE_6to4=y
CONFIG_PACKAGE_adbyby=y
CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_etherwake=y
CONFIG_PACKAGE_ip6tables=y
CONFIG_PACKAGE_ipv6helper=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-iptunnel4=y
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-sit=y
CONFIG_PACKAGE_libminiupnpc=y
CONFIG_PACKAGE_libnatpmp=y
CONFIG_PACKAGE_libstdcpp=y
# CONFIG_PACKAGE_luci-app-nlbwmon is not set
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Libev_Client is not set
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Xray=y
# CONFIG_PACKAGE_luci-app-ttyd is not set
# CONFIG_PACKAGE_luci-app-unblockmusic is not set
# CONFIG_PACKAGE_luci-app-vlmcsd is not set
# CONFIG_PACKAGE_luci-app-vsftpd is not set
CONFIG_PACKAGE_luci-app-watchcat=y
CONFIG_PACKAGE_luci-app-wolplus=y
CONFIG_PACKAGE_luci-app-zerotier=y
CONFIG_PACKAGE_luci-i18n-watchcat-zh-cn=y
CONFIG_PACKAGE_luci-i18n-wolplus-zh-cn=y
CONFIG_PACKAGE_luci-i18n-zerotier-zh-cn=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcp6c_ext_cer_id=0
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_odhcpd_ipv6only_ext_cer_id=0
CONFIG_PACKAGE_watchcat=y
CONFIG_PACKAGE_xray-core=y
CONFIG_PACKAGE_zerotier=y
CONFIG_PACKAGE_kmod-nf-conntrack-netlink=y
CONFIG_PACKAGE_libcap=y
# CONFIG_PACKAGE_libcap-bin is not set
CONFIG_PACKAGE_libwebsockets-full=y
CONFIG_PACKAGE_nlbwmon=y
CONFIG_PACKAGE_shadowsocksr-libev-ssr-local=y
CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_vlmcsd=y
CONFIG_PACKAGE_vsftpd-alt=y
CONFIG_VSFTPD_USE_UCI_SCRIPTS=y`

281677160, 2022-04-23:
翻墙插件只选主程序，下面的子选项一个都不要选

shelken, 2022-04-25:
> 

不可以，我的机场是 vmess 协议，必须 要v2ray；我也只在 menu 中选了一个 v2ray 核心。

shelken, 2022-04-25:
CONFIG_PACKAGE_shadowsocksr-libev-ssr-local=y
CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir=y
我也不太理解为什么这两行会出现，我是取消勾选的。

drw2005, 2022-04-25:
和我编译小米mini情况一样，恩山上有些发固件的，插件多很多，体积也才13M左右，我就选个SSR+PLUS，就主体+xary，就超过15M了，甚至大部分时侯直接出不来固件。
一样困惑是哪里不对，网上找了好久，始终找不到答案。

drw2005, 2022-04-27:
> 翻墙插件只选主程序，下面的子选项一个都不要选

子选项一个都不选，体积确实小了很多。但是SS/SSR/V2RAY/TROJAN之类的节点都不能用，直接ShadowsocksR Plus+ NOT RUNNING了。请教大佬是啥原因？

281677160, 2022-04-27:
> > 翻墙插件只选主程序，下面的子选项一个都不要选
> 
> 子选项一个都不选，体积确实小了很多。但是SS/SSR/V2RAY/TROJAN之类的节点都不能用，直接ShadowsocksR Plus+ NOT RUNNING了。请教大佬是啥原因？

不了解，老早之前就叫人测试了，是没问题的，X86的我刚测试了一下也是没问题的

drw2005, 2022-04-27:
> > > 翻墙插件只选主程序，下面的子选项一个都不要选
> > 
> > 
> > 子选项一个都不选，体积确实小了很多。但是SS/SSR/V2RAY/TROJAN之类的节点都不能用，直接ShadowsocksR Plus+ NOT RUNNING了。请教大佬是啥原因？
> 
> 不了解，老早之前就叫人测试了，是没问题的，X86的我刚测试了一下也是没问题的

刚才重新编译，选了下面的Include Shadowsocks Libev Client子选项就可以了，光主程序不行。

281677160, 2022-04-27:
> Include Shadowsocks Libev Client

呃，好像是改过了，各个机型有点不一样了
[image]
[image]
第一个图片的就是只有主程序的，啥节点都没问题，下面图片是另外一个机型，只选主程序，但是少了好几个，我看了好几个小删除的机型，大雕的默认就是主程序+Include Shadowsocks Libev Client子选项了

drw2005, 2022-04-28:
> > Include Shadowsocks Libev Client
> 
> 呃，好像是改过了，各个机型有点不一样了 [image] [image] 第一个图片的就是只有主程序的，啥节点都没问题，下面图片是另外一个机型，只选主程序，但是少了好几个，我看了好几个小删除的机型，大雕的默认就是主程序+Include Shadowsocks Libev Client子选项了

问题是光主程序+Include Shadowsocks Libev Client子选项，没办法支持象TROJA、V2RAY和VMESS之类的节点，只能SS、SSR节点能用。

closed 2022-05-13 by 281677160