281677160/build-actions #60: 脚本需要更新,报错luci-app-cupsd/Makefile&&luci-app-mosdns/Makefile

lee29, 2022-06-27:
`Collecting target info: doneWARNING: Makefile 'package/feeds/danshui/luci-app-cupsd/Makefile' has a dependency on 'luci-lib-ipkg', which does not exist
WARNING: Makefile 'package/feeds/danshui/luci-app-mosdns/Makefile' has a dependency on 'mosdns_neo', which does not exist`

目前只能这样 啦
        sed -i 's/mosdns_neo/mosdns/g' ./package/feeds/danshui/luci-app-mosdns/Makefile
        sudo rm -rf ./package/feeds/danshui/luci-app-cupsd
        sudo rm -rf ./feeds/danshui/luci-app-cupsd

281677160, 2022-07-06:
问题已解决了，我删除了我插件包的这2个插件了

lee29, 2022-07-06:
可以保留,但要加个命令
`     sudo rm -rf ./feeds/packages/net/mosdns && cp -rf ./feeds/kenzo/mosdns ./feeds/packages/net/mosdns
     sed -i 's/mosdns_neo/mosdns/g' ./feeds/kenzo/luci-app-mosdns/Makefile
     sed -i 's/mosdns_neo/mosdns/g' ./feeds/kenzo/mosdns/Makefile
     sed -i 's/mosdns_neo/mosdns/g' ./feeds/packages/net/mosdns/Makefile`