281677160/build-actions #72: ubuntu20.04本地编译出错。。。

drw2005, 2022-10-17:
编译x86固件，大雕的源码没问题，其他的Lienol和 immortalwrt都出现下面的错误。

> find /home/drw/openwrt/build_dir/target-i386_pentium4_musl/root-x86/ -mindepth 1 -execdir touch -hcd "@1664995310" "{}" +
find: The relative path 'Files' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find. Please remove that entry from $PATH
make[2]: *** [package/Makefile:73: package/install] Error 1
make[2]: Leaving directory '/home/drw/openwrt'
make[1]: *** [package/Makefile:111: /home/drw/openwrt/staging_dir/target-i386_pentium4_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/drw/openwrt'
make: *** [/home/drw/openwrt/include/toplevel.mk:230: world] Error 2

281677160, 2022-10-17:
那么点日志看毛线，你重新安装ubuntu看看吧，还不行就用22.02测试一下，我早就不搞这些了，云编译测试了一下是没问题的

drw2005, 2022-10-18:
> find: The relative path 'Files' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find. Please remove that entry from $PATH

ubuntu也重新安装了，依赖也都重新安装了，应该不是这个原因。一直提示：find: The relative path 'Files' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find. Please remove that entry from $PATH。
有没有解决办法？

closed 2022-10-21 by drw2005