281677160/build-actions #73: ubuntu22.04本地编译Lienol源码出不来固件。。。

drw2005, 2022-10-21:
> 2022年10月9号，因为上游Lienol大佬删除了22.03源码，我这里也就把22.03改成他的21.02源码了，编译脚本的ubuntu也从20.04改成22.04
LEDE源码编译正常，但是这两天用Lienol大佬的源码编译，最后都出不来固件，最后一步提示：找不到firmware目录。
immortalwrt源码编译也是如此。
[image]

另外，还有这个提示：
[image]

281677160, 2022-10-22:
kmod-inet-diag错误不影响编译的，miniupnp编译有错误，肯定出不来固件啊

drw2005, 2022-10-22:
> miniupnp

请问miniupnp错误是什么原因？

drw2005, 2022-10-28:
> kmod-inet-diag错误不影响编译的，miniupnp编译有错误，肯定出不来固件啊

在大雕那里看到有人说，是因为和windows的环境变量混乱导致的。
有人给出解决办法：
在make前面加上：
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
比如：
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin make

能看看给加上不？我这边到时再试试。

281677160, 2022-10-28:
脚本里面不是本来就有？

drw2005, 2022-10-28:
> 脚本里面不是本来就有？
> […](#)

我的报错和这个一样的：
https://github.com/coolsnowwolf/lede/issues/3822

drw2005, 2022-11-01:
禁用windows环境就可以编译成功了，但是都是用的coolsnowwolf、Lienol、immortalwrt三位的原始命令编译成功的。

用老大你的依然只能到checksum处，然后提示：无法访问，没有那个文件或目录。

closed 2022-11-15 by drw2005