281677160/build-actions #75: lede 编译gost会报错了！~

czyjian, 2022-11-29:
lede 编译gost会报错了！~

czyjian, 2022-11-29:
Building targets
github.com/lucas-clemente/quic-go/internal/qtls
# github.com/lucas-clemente/quic-go/internal/qtls
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.26.0/internal/qtls/go119.go:6:13: cannot use "quic-go doesn't build on Go 1.19 yet." (untyped string constant) as int value in variable declaration

make[3]: *** [Makefile:50: /Lede_source/openwrt/build_dir/target-x86_64_musl/gost-2.11.2/.built] Error 2
make[3]: Leaving directory '/Lede_source/openwrt/feeds/danshui/gost'
time: package/feeds/danshui/gost/compile#0.54#0.27#0.64
    ERROR: package/feeds/danshui/gost failed to build.
make[2]: *** [package/Makefile:116: package/feeds/danshui/gost/compile] Error 1
make[2]: Leaving directory '/Lede_source/openwrt'
make[1]: *** [package/Makefile:110: /Lede_source/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Lede_source/openwrt'
make: *** [/Lede_source/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

281677160, 2022-12-12:
我拉回老一点的版本了，刚编译了一次，没问题，你再试试看

closed 2023-02-16 by 281677160