281677160/build-actions #83: 使用官方的mac80211源码替换Lede，无线网络异常

Luupw3d, 2023-04-01:
首先感谢作者的贡献，Lede大雕源码在去年某一时候便出现了mac80211编译出错(ath10k-ct报错误02)的问题，首次在此看到使用官方源码替换的方法，故进行尝试。

使用官方mac80211源码替换后，竞斗云固件的确可以编译成功，但无线无法正常使用，Kernel log如下：
> [   30.235190] ath10k_core: unknown parameter 'frame_mode' ignored
[   40.313209] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   40.313567] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   40.372667] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   40.419976] ath10k_ahb a000000.wifi: Loading BDF type 0
[   40.420506] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 6ce1f887
[   41.711070] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   41.711115] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   41.747086] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   41.748040] ath10k_ahb a000000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
[   41.878086] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   42.038964] ath: EEPROM regdomain: 0x0
[   42.039037] ath: EEPROM indicates default country code should be used
[   42.039083] ath: doing EEPROM country->regdmn map search
[   42.039177] ath: country maps to regdmn code: 0x3a
[   42.039241] ath: Country alpha2 being used: US
[   42.039293] ath: Regpair used: 0x3a
[   44.638558] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   44.638605] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   44.650722] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 6b2b5c5b
[   44.702283] ath10k_ahb a800000.wifi: Loading BDF type 0
[   44.702611] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:17 crc32 6ce1f887
[   45.993857] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   45.993901] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   46.029678] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   46.030631] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13432 sram: 35752'
[   46.157855] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   46.309908] ath: EEPROM regdomain: 0x0
[   46.309983] ath: EEPROM indicates default country code should be used
[   46.310032] ath: doing EEPROM country->regdmn map search
[   46.310129] ath: country maps to regdmn code: 0x3a
[   46.310192] ath: Country alpha2 being used: US
[   46.310243] ath: Regpair used: 0x3a

这个是否还有其他解决方法呢？

281677160, 2023-04-02:
你直接把他自带源码的错误日志拿去找源码作者不是更好？源码作者都不能修复，我也没办法

Luupw3d, 2023-04-02:
> 你直接把他自带源码的错误日志拿去找源码作者不是更好？源码作者都不能修复，我也没办法

本issue主要为mac80211替换修复方法做了个测试，证明目前该方法是可编译、不可用的，可以在脚本说明中做出提示。至于该问题，将向Lede项目反馈。谢谢作者！

closed 2023-04-02 by Luupw3d