281677160/build-actions #87: 天灵报错

lj0307222, 2023-04-24:
云编译开始编译就出现，小白
Run cd openwrt
WARNING: Makefile 'package/emortal/default-settings/Makefile' has a dependency on 'luci-i18n-base-zh_Hans', which does not exist

281677160, 2023-04-24:
去找天灵反映，他源码的default-settings增加luci-i18n-base-zh-cn造成的，我可以在这里修复一下，不过没啥必要，跟天灵反映问题，一般很快可以解决

lj0307222, 2023-04-24:
好的，感谢大佬的付出

lj0307222, 2023-04-24:
好的，感谢大佬的付出

lj0307222, 2023-04-24:
> 去找天灵反映，他源码的default-settings增加luci-i18n-base-zh-cn造成的，我可以在这里修复一下，不过没啥必要，跟天灵反映问题，一般很快可以解决

[image]
大佬这。。😅😅

281677160, 2023-04-24:
一头雾水，等我有时间在看看吧，我记得好像没修改这文件啊

281677160, 2023-04-24:
[image]

这个修改插件语言的文件，把makefile的数值给改了，我也郁闷老半天，也找了老半天才发现这个问题的，哈哈，你去跟天灵说声不好意思吧，特么的这个真是意外

运行这个文件也是有用的，比如你外加插件的时候，插件本身不是zh_Hans格式的语言包，就不会显示中文了，要这个文件来修改一下

waynesg, 2023-04-24:
我试过把这个语言转换脚本禁用，会报   ERROR: package/base-files failed to build.错误。

281677160, 2023-04-24:
> 我试过把这个语言转换脚本禁用，会报 ERROR: package/base-files failed to build.错误。

这个脚本跟其他没啥关系啊，不想用就不要用呗，就是修改插件成中文的，以前的插件都是18.06或者17.01LUCI的，可以说成是LUCI吧，中文语言包都是 zh-cn 的，后面的LUCI2 都是zh_Hans ,这个文件就是修改所有 zh-cn成zh_Hans

closed 2023-09-11 by 281677160