281677160/build-actions #90: 编译成功却没有无线信号

leegyack, 2023-05-22:
请教先生一个问题：
我编译CM520-0795固件，用Lede系统，设定了固件类型以后，从Luci-Aplication选择插件，编译成功，从github可以下载固件，但是没有发射无线信号，在wifi表里面也不显示ssid。
查看了你的seed，在lede/x86-64是这样的，
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
而我的前三行还是老的：
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_DEVICE_mobipromo_cm520-79f=y
CONFIG_TARGET_BOARD="ipq40xx"
不知道是不是缺省设置有了改变？应该在哪里设置无线？
谢谢！

281677160, 2023-05-22:
你没选网卡驱动吧，无线一般都是mac80211

leegyack, 2023-05-22:
我想不起来该在哪里选择网卡，请先生指教一下，好吗？

leegyack, 2023-05-23:
谢谢指教，putty很难接上ssh，多半用网页，受时间限制，只在luci/application菜单上匆忙选择，很难仔细查找。

leegyack, 2023-08-06:
上次反映的问题未能够解决，只好退回老固件，现在看见有更新了，又新做了一个18.06 LEDE的CM520-73F的固件，查了一下Kernel modules --->Wireless Drivers --->里面mac80211是缺省选择，但是更新固件以后，以LAN线接通后台，有无线显示，但是没有发射无线信号，SSID不显示，两个wan灯5.8g与2.4g都不亮，无线没法连接。查了一下，这款路由器的网卡芯片是QCA8075，不知道是不是有特定的驱动，或者是mac80211的驱动还存在问题，烦请指教一下。谢谢。

281677160, 2023-08-12:
自己到大雕源码的IS那里问问吧

closed 2023-09-11 by 281677160