281677160/build-actions #91: x86编译失败

yezivvvv, 2023-06-03:
最近一直编译失败，超过六个小时自动取消。配置没有改变以前都可以编译成功。
可以帮忙看看什么问题吗？

g++  -o /Ledemaster/openwrt/build_dir/target-x86_64_musl/node-v16.19.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/wasm/wasm-debug.o ../deps/v8/src/wasm/wasm-debug.cc '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DV8_TARGET_ARCH_X64' '-DV8_HAVE_TARGET_OS' '-DV8_TARGET_OS_LINUX' '-DV8_EMBEDDER_STRING="-node.24"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_INTL_SUPPORT' '-DV8_ATOMIC_OBJECT_FIELD_WRITES' '-DV8_ATOMIC_MARKING_STATE' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DV8_ENABLE_WEBASSEMBLY' '-DV8_ALLOCATION_FOLDING' '-DV8_ALLOCATION_SITE_TRACKING' '-DV8_ADVANCED_BIGINT_ALGORITHMS' '-DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_STATIC' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_STATIC_IMPLEMENTATION=1' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' -I../deps/v8 -I../deps/v8/include -I/Ledemaster/openwrt/build_dir/target-x86_64_musl/node-v16.19.1/out/Release/obj/gen/inspector-generated-output-root -I../deps/v8/third_party/inspector_protocol -I/Ledemaster/openwrt/build_dir/target-x86_64_musl/node-v16.19.1/out/Release/obj/gen -I/Ledemaster/openwrt/build_dir/target-x86_64_musl/node-v16.19.1/out/Release/obj/gen/generate-bytecode-output-root -I../deps/icu-small/source/common -I../deps/icu-small/source/i18n -I../deps/icu-small/source/tools/toolutil -I../deps/v8/third_party/zlib -I../deps/v8/third_party/zlib/google -I/Ledemaster/openwrt/staging_dir/target-x86_64_musl/usr/include  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++14 -MMD -MF /Ledemaster/openwrt/build_dir/target-x86_64_musl/node-v16.19.1/out/Release/.deps//Ledemaster/openwrt/build_dir/target-x86_64_musl/node-v16.19.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/wasm/wasm-debug.o.d.raw   -c
Error: The operation was canceled.

281677160, 2023-06-03:
你这样神仙都帮不了你，编译失败就下载日志查看，你说你看不懂，发过来看看还有机会看明白，虽然好多错误我自己都看不懂

yezivvvv, 2023-06-03:
去找了日志看了，百度了下这个错误是网络出问题了？

2023-06-03T07:19:41.9900372Z  make[3] -C feeds/luci/applications/luci-app-ttyd compile
2023-06-03T07:19:44.4540598Z  make[3] -C feeds/danshui2/luci-app-argon-config compile
2023-06-03T07:19:47.4260438Z  make[3] -C feeds/danshui2/luci-app-design-config compile
2023-06-03T07:19:50.0901516Z  make[3] -C feeds/luci/collections/luci compile
2023-06-03T07:19:51.7901237Z  make[3] -C feeds/luci/applications/luci-app-autoreboot compile
2023-06-03T07:19:54.9580446Z  make[3] -C package/lean/default-settings compile
2023-06-03T09:07:22.8568744Z  make[3] -C feeds/danshui1/luci-app-unblockneteasemusic compile
2023-06-03T09:07:23.4755221Z  make[2] package/install
2023-06-03T09:07:44.9128761Z make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
2023-06-03T09:07:44.9212808Z make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 1
2023-06-03T09:07:54.9574280Z make[2]: Entering directory '/Ledemaster/openwrt/scripts/config'
2023-06-03T09:07:54.9575574Z make[2]: 'conf' is up to date.
2023-06-03T09:07:54.9577660Z make[2]: Leaving directory '/Ledemaster/openwrt/scripts/config'
2023-06-03T09:07:57.2645736Z make[1]: Entering directory '/Ledemaster/openwrt'
2023-06-03T09:07:59.9269913Z make[2]: Entering directory '/Ledemaster/openwrt'
2023-06-03T09:08:02.5352002Z make[3]: Entering directory '/Ledemaster/openwrt/target/linux'
2023-06-03T09:08:02.5525336Z make[4]: Entering directory '/Ledemaster/openwrt/target/linux/x86'
2023-06-03T09:08:02.6300964Z rm -f /Ledemaster/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.31/localversion

281677160, 2023-06-03:
有可能是luci-app-unblockneteasemusic这个插件错误了，取消这个插件还有相关的依赖再编译看看

yezivvvv, 2023-06-05:
还是有问题

yezivvvv, 2023-06-07:
应该是某个插件的问题，我用默认配置编译成功了。

closed 2023-06-07 by yezivvvv

281677160, 2023-06-07:
看日志啊，不知道哪里错，怎么搞