281677160/build-actions #96: 如何将seed文件还原到初始值

leegyack, 2023-08-27:
由于编译的cm520-79f固件无线没有发射信号，蒙先生指示上大雕仓库查看issues,同样问题不少，大概都是Kernel modules/Wireless Drivers /下面的ath10k-ct上有问题，查了一下21年10月的固件的manifest是这样的：
2021年10月26日
--------------------------------------------
ath10k-board-qca4019 - 20210511-1
ath10k-firmware-qca4019-ct - 2020-11-08-1
---------------------------------------------------------
kmod-ath - 5.4.155+5.10.42-1-3
kmod-ath10k-ct - 5.4.155+2021-06-03-b44cd7b2-0
kmod-cfg80211 - 5.4.155+5.10.42-1-3
---------------------------------------------
kmod-mac80211 - 5.4.155+5.10.42-1-3

而2023年8月的固件是：
2023年8月21日

ath10k-board-qca4019 - 20230625-1
ath10k-firmware-qca4019-ct - 2020-11-08-1
-------------------------------------------------------
kmod-ath - 5.4.254+6.1.24-3
kmod-ath10k-ct - 5.4.254+2022-05-13-f808496f-3
kmod-cfg80211 - 5.4.254+6.1.24-3
-------------------------------------------------
kmod-mac80211 - 5.4.254+6.1.24-3
请问，能否将以上插件退回到2021年老版本，如可能该如何操作？
另外，由于加了hysteria，编译一直出错，想将其去除，结果将seed弄乱了，无法修改，如何将其还原到x86-64的三行初始状态，请指教。

281677160, 2023-08-27:
x86的直接清空配置文件就可以了

DanShui
***@***.***

leegyack, 2023-08-29:
我已经试过了，需要拉取patch-file，每次到这里都出错，删除x86-64也不行，我只有重新拉仓库。但是忘记如何安放repo_token，查先生的教材《仓库密匙获取跟使用》，里面的图都不可见了。希望予以指教。

281677160, 2023-08-29:
> 我已经试过了，需要拉取patch-file，每次到这里都出错，删除x86-64也不行，我只有重新拉仓库。但是忘记如何安放repo_token，查先生的教材《仓库密匙获取跟使用》，里面的图都不可见了。希望予以指教。

图片看不到，那是你网络问题

closed 2023-09-11 by 281677160