281677160/build-actions #97: 最近几次编译都失败了，大佬看看错在哪里。。。

drw2005, 2023-08-28:
上星期还一切正常

install -m0755 ./files/sysntpd /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/etc/init.d/sysntpd
install -m0755 ./files/ntpd-hotplug /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/usr/sbin/ntpd-hotplug
install -d -m0755 /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/etc/capabilities /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/usr/share/acl.d
install -m0644 ./files/ntpd.capabilities /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/etc/capabilities/ntpd.json
install -m0644 ./files/ntpd_acl.json /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/usr/share/acl.d/ntpd.json
rm -rf /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/lib64
find /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/drw/openwrt/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/drw/openwrt/scripts/strip-kmod.sh" PATCHELF="/home/drw/openwrt/staging_dir/host/bin/patchelf" /home/drw/openwrt/scripts/rstrip.sh /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox
rstrip.sh: /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/bin/busybox: executable
(cd /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm; echo "$V_Package_busybox_conffiles" > conffiles;  )
install -d -m0755 /home/drw/openwrt/bin/packages/x86_64/base
/home/drw/openwrt/staging_dir/host/bin/fakeroot /home/drw/openwrt/staging_dir/host/bin/bash /home/drw/openwrt/scripts/ipkg-build -m "" /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox /home/drw/openwrt/bin/packages/x86_64/base
Packaged contents of /home/drw/openwrt/build_dir/target-x86_64_musl/busybox-default/busybox-1.36.0/ipkg-x86_64/busybox into /home/drw/openwrt/bin/packages/x86_64/base/busybox_1.36.0-1_x86_64.ipk
echo "busybox" >> /home/drw/openwrt/staging_dir/target-x86_64_musl/pkginfo/busybox.default.install
make[3]: Leaving directory '/home/drw/openwrt/package/utils/busybox'
time: package/utils/busybox/default/compile#68.96#7.38#38.01
make[2]: Leaving directory '/home/drw/openwrt'
make[1]: *** [package/Makefile:110: /home/drw/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/drw/openwrt'
make: *** [/home/drw/openwrt/include/toplevel.mk:231: world] Error 2

281677160, 2023-09-01:
本地编译的话，把源码删除了重新拉取再试试

drw2005, 2023-09-01:
> 本地编译的话，把源码删除了重新拉取再试试

当时删除重新拉取也不行，估计源码哪里有问题，昨天重新拉取可以了。

closed 2023-09-01 by drw2005