281677160/build-actions #103: 官方源码-23.05分支，插件更新失败

a790186435, 2023-09-23:
官方源码-23.05分支，插件更新失败

Collecting target info: donesh -c 'cat '\''/home/runner/work/OpenWrt/OpenWrt/openwrt/build/Official/patches/test.patch'\''  | patch -d '\''./'\'' -p1 --forward --no-backup-if-mismatch'
patching file package/kernel/mt76/Makefile
Hunk #1 succeeded at 278 with fuzz 1 (offset 16 lines).
Hunk #2 succeeded at 605 (offset 66 lines).
Hunk #3 FAILED at 580.
1 out of 3 hunks FAILED -- saving rejects to file package/kernel/mt76/Makefile.rej
Error: Process completed with exit code 123.

281677160, 2023-09-23:
写了啊，build/Official/patches/test.patch补丁有问题
自己弄了啥自己搞定

closed 2023-09-23 by 281677160

a790186435, 2023-09-23:
懂了，大版本更新原来只是覆盖文件呀，我以为所有文件都被移到backups了，就没再去检查build