281677160/build-actions #110: passwall插件出错

hbzmt, 2023-11-29:
make[3]: *** [Makefile:91: /Ledemaster/openwrt/build_dir/target-x86_64_musl/Xray-core-1.8.6/.built] Error 1
make[3]: Leaving directory '/Ledemaster/openwrt/feeds/passwall3/xray-core'
time: package/feeds/passwall3/xray-core/compile#0.12#0.13#0.22
    ERROR: package/feeds/passwall3/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/feeds/passwall3/xray-core/compile] Error 1
make[2]: Leaving directory '/Ledemaster/openwrt'
make[1]: *** [package/Makefile:110: /Ledemaster/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Ledemaster/openwrt'
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

ljkk-li, 2023-12-03:
大佬，快来更新更新啊，我也是

hbzmt, 2024-01-02:
> 呃，好久不玩这个了，现在叫我修，我也不知道从那里下手了，看看啥时候有空，弄一个不带插件的仓库吧，不带插件的没那么多问题出现

大佬，之前一直用你这个的，lean源码编译的，很好用，autoupdate一直稳定运行，后来你这个不能编译之后，换了好几个别人的版本，都不行，无法自动升级。麻烦你抽空更新一下，我感觉问题出在你拉取的ssr或者passwall上，每次到x-ray核心编译这里都出问题。试图自己去添加一个源也不行。SSH连接，取消选择ssr和passwall也不行。实在是毫无头绪。

hbzmt, 2024-01-02:
> 呃，好久不玩这个了，现在叫我修，我也不知道从那里下手了，看看啥时候有空，弄一个不带插件的仓库吧，不带插件的没那么多问题出现

我感觉是拉取的packages出问题了，你可以看下这个https://github.com/roacn/openwrt-packages，我感觉蛮全的，而且更新速度也还不错，我用他的build-actions可以编译成功的，但是自动更新有问题，使用ygkkk的优选IP脚本停止PASSWALL后无法启动。感觉还是整体固件有问题。还是觉得你的这个比较好，烦请有空修复一下，万分感谢。

hbzmt, 2024-01-02:
> 呃，好久不玩这个了，现在叫我修，我也不知道从那里下手了，看看啥时候有空，弄一个不带插件的仓库吧，不带插件的没那么多问题出现

大佬或者麻烦告诉一下，你这个拉取仓库在哪里修改，我试着自己去弄下，我找了很久，没找到你的拉取仓库脚本在哪里，我对于编译不是很精通。

281677160, 2024-01-04:
golang版本引起的错误，现在应该没问题了，源码作者基本都更新了最新的golang版本

hbzmt, 2024-01-07:
> golang版本引起的错误，现在应该没问题了，源码作者基本都更新了最新的golang版本
已经编译成功,非常感谢大佬修复.

ljkk-li, 2024-01-16:
感谢大佬百忙之中修复仓库