281677160/build-actions #111: 不能正常编译了，大佬有时间修一下吧。。。

drw2005, 2023-12-18:
好久没编译了，今天发现已经不能正常编译了，大佬有时间修一下吧。
很多人在用的，别放弃啊。

zhy668, 2023-12-20:
期待大佬更新

tom2025, 2023-12-22:
别放弃，别放弃，别放弃。

EdisonCh4n, 2023-12-23:
期待期待！

John-youn-master, 2023-12-25:
期待大佬更新，先更新lede的也可以啊

zuifengguai, 2023-12-26:
我还以为是插件冲突问题，原来都编译不起了所

chdeast, 2023-12-29:
官方版22.03编译失败，tailscale插件失败；23.05正常。非常好用的编译库，期待您的更新！

281677160, 2024-01-04:
golang版本引起的错误，现在应该没问题了，源码作者基本都更新了最新的golang版本

chdeast, 2024-01-05:
非常感谢。

281677160, 2024-01-05:
昨晚上测试了所有源码及分支了，都没啥问题了

281677160, 2024-01-05:
> 官方版22.03编译失败，tailscale插件失败；23.05正常。非常好用的编译库，期待您的更新！

你这个插件错误，我不知道你说的是哪个，你现在去编译测试看看吧，日志也没有，啥都不清楚

chdeast, 2024-01-08:
> > 官方版22.03编译失败，tailscale插件失败;23.05正常。非常好用的编译库，期待您的更新！
> 
> 你这个插件错误，我不知道你说的是哪个，你现在去编译测试看看吧，日志也没有，啥都不清楚

感谢您的回复。我又重新了官方22.03-x86版本，还是失败了。附上日志，供您分析，谢谢。
make[3]: Leaving directory '/Official2203/openwrt/feeds/packages/net/openssh'
time: package/feeds/packages/openssh/without-pam/compile#0.17#0.07#0.22
make[3]: Entering directory '/Official2203/openwrt/feeds/packages/net/tailscale'
rm -f /Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3/.built
touch /Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/Official2203/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/Official2203/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="tailscale.com/cmd/tailscale tailscale.com/cmd/tailscaled" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="tailscale.com/cmd/tailscale tailscale.com/cmd/tailscaled/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3" GO_BUILD_DIR="/Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3/.go_work/build" GO_BUILD_BIN_DIR="/Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/Official2203/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3=tailscale-1.32.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/include -I/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include/fortify -I/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3=tailscale-1.32.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -znow -zrelro" GOPATH="/Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3/.go_work/build" GOCACHE="/Official2203/openwrt/tmp/go-build" GOMODCACHE="/Official2203/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1704403786' -linkmode external -extldflags '-L/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-X 'tailscale.com/version.Long=1.32.3-1 (OpenWrt)' -X tailscale.com/version.Short=1.32.3 -buildid '1704403786' -linkmode external -extldflags '-L/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/Official2203/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets
package tailscale.com/cmd/tailscaled
	imports tailscale.com/ipn/ipnserver
	imports tailscale.com/ipn/ipnlocal
	imports tailscale.com/net/dns
	imports tailscale.com/net/tstun
	imports gvisor.dev/gvisor/pkg/tcpip
	imports gvisor.dev/gvisor/pkg/atomicbitops
	imports gvisor.dev/gvisor/pkg/state
	imports gvisor.dev/gvisor/pkg/state/wire
	imports gvisor.dev/gvisor/pkg/gohacks: build constraints exclude all Go files in /Official2203/openwrt/dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20220817001344-846276b3dbc5/pkg/gohacks

Building targets
package tailscale.com/cmd/tailscaled
	imports tailscale.com/ipn/ipnserver
	imports tailscale.com/ipn/ipnlocal
	imports tailscale.com/net/dns
	imports tailscale.com/net/tstun
	imports gvisor.dev/gvisor/pkg/tcpip
	imports gvisor.dev/gvisor/pkg/atomicbitops
	imports gvisor.dev/gvisor/pkg/state
	imports gvisor.dev/gvisor/pkg/state/wire
	imports gvisor.dev/gvisor/pkg/gohacks: build constraints exclude all Go files in /Official2203/openwrt/dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20220817001344-846276b3dbc5/pkg/gohacks

make[3]: *** [Makefile:86: /Official2203/openwrt/build_dir/target-x86_64_musl/tailscale-1.32.3/.built] Error 1
make[3]: Leaving directory '/Official2203/openwrt/feeds/packages/net/tailscale'
time: package/feeds/packages/tailscale/compile#0.59#0.43#0.65
    ERROR: package/feeds/packages/tailscale failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/tailscale/compile] Error 1
make[2]: Leaving directory '/Official2203/openwrt'
make[1]: *** [package/Makefile:114: /Official2203/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Official2203/openwrt'
make: *** [/Official2203/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

281677160, 2024-01-09:
> tailscale

tailscale版本过低吧，我把23.05的换过去了，你自己测试一下行不行吧

closed 2024-01-14 by 281677160