281677160/build-actions #113: 大佬，修过后又出现编译错误了

alexhou83, 2024-01-13:
make[4]: *** [Makefile:1226: tools/chacl.o] Error 1
make[4]: Leaving directory '/Ledemaster/openwrt/build_dir/target-x86_64_musl/acl-2.3.1'
make[3]: *** [Makefile:87: /Ledemaster/openwrt/build_dir/target-x86_64_musl/acl-2.3.1/.built] Error 2
make[3]: Leaving directory '/Ledemaster/openwrt/feeds/packages/utils/acl'
time: package/feeds/packages/acl/compile#0.16#0.06#0.20
    ERROR: package/feeds/packages/acl failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/acl/compile] Error 1
make[2]: Leaving directory '/Ledemaster/openwrt'
make[1]: *** [package/Makefile:110: /Ledemaster/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Ledemaster/openwrt'
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

不清楚哪里错了，麻烦帮忙看下，谢谢！

281677160, 2024-01-13:
大雕8小时前修复过acl了，你自己再编译看看吧

closed 2024-01-14 by 281677160

hbzmt, 2024-01-16:
> 大雕8小时前修复过acl了，你自己再编译看看吧
谢谢，已经成功。