281677160/build-actions #118: ERROR: package/feeds/packages/aliyundrive-webdav failed to build

tom2025, 2024-01-17:
ERROR: package/feeds/packages/aliyundrive-webdav failed to build.
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 1

281677160, 2024-01-18:
在diy-part.sh文件加入以下代码看看
````
curl -o ./feeds/packages/lang/rust/rust-values.mk https://raw.githubusercontent.com/Jason6111/packages/patch-1/lang/rust/rust-values.mk
curl -o ./feeds/packages/lang/rust/Makefile https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/Makefile
curl -o ./feeds/packages/lang/rust/patches/0001-Update-xz2-and-use-it-static.patch https://raw.githubusercontent.com/Jason6111/OpenWrt_Personal/main/other/rust/0001-Update-xz2-and-use-it-static.patch
````

281677160, 2024-01-19:
> 添加代码了，又出现警告了。 Run cd openwrt 4 thread compile WARNING: Makefile 'package/feeds/packages/aliyundrive-webdav/Makefile' has a build dependency on 'rust/host', which does not exist WARNING: Makefile 'package/feeds/packages/maturin/Makefile' has a build dependency on 'rust/host', which does not exist WARNING: Makefile 'package/feeds/packages/ripgrep/Makefile' has a build dependency on 'rust/host', which does not exist make[2]: Entering directory '/Ledemaster/openwrt/scripts/config' make[2]: 'conf' is up to date. make[2]: Leaving directory '/Ledemaster/openwrt/scripts/config' make[1] world

把前面的命令去掉，用这个看看吧，还不行就等大雕修复吧
````
curl -o ./feeds/packages/multimedia/aliyundrive-webdav/Makefile https://raw.githubusercontent.com/coolsnowwolf/packages/aea60b5432fad984c0a4013bad0f0c5e00dcd115/multimedia/aliyundrive-webdav/Makefile
````

tom2025, 2024-01-20:
编译成功了，但是无论是升级安装还是全新安装都无法进系统，真的很奇怪！

[image]

tom2025, 2024-01-20:
固件地址如下，大家可以测试一下是什么问题。
https://www.123pan.com/s/KwUSVv-3ZOCh.html

281677160, 2024-01-20:
> 固件地址如下，大家可以测试一下是什么问题。 https://www.123pan.com/s/KwUSVv-3ZOCh.html

没有支持UEFI的设备测试，你还是换一个作者的编译工具玩吧

closed 2024-01-20 by 281677160

tom2025, 2024-01-22:
找到原因了，RTL8153 USB网卡驱动不起来导致进不了系统，只能等下个版本了。