281677160/build-actions #126: 编译LEDE固件默认是18.06？CPU优选可以选择其它吗？

xlfq3136, 2024-02-15:
1、workflow里大雕源码明明是master
    inputs:
      REPO_BRANCH:
        description: '大雕源码'
        required: true
        **default: 'master'**
        type: choice
        options:
          - 'master'
为什么编译的时候是18.06分支？（弃用E5-编译-Lede-18.06-hiwifi_hc5962固件）

2、再求问一下CPU优选编译选项可以改成8370C而非默认吗？见下：
CPU_SELECTION:
        description: 'CPU优选,增加编译速度'
        required: true
        **default: '默认'**
        type: choice
        options:
          - '默认'
          - '8370C'
          - '8272CL'
          - '8171M'

281677160, 2024-02-16:
大雕的LUCI版本一直都是18.06啊，这个是LUCI版本而已
CPU选择现在好像基本没用了，这几天测试脚本的时候看到的全部是AMD EPYC 7763的CPU，其他CPU都没见过，如果你不搞默认的话，有可能选择不了CPU而一直寻找CPU，无限死循环

xlfq3136, 2024-02-16:
> 大雕的LUCI版本一直都是18.06啊，这个是LUCI版本而已
> CPU选择现在好像基本没用了，这几天测试脚本的时候看到的全部是AMD EPYC 7763的CPU，其他CPU都没见过，如果你不搞默认的话，有可能选择不了CPU而一直寻找CPU，无限死循环

求教大雕的其它版本如何编译啊？

281677160, 2024-02-16:
> > 大雕的LUCI版本一直都是18.06啊，这个是LUCI版本而已
> > CPU选择现在好像基本没用了，这几天测试脚本的时候看到的全部是AMD EPYC 7763的CPU，其他CPU都没见过，如果你不搞默认的话，有可能选择不了CPU而一直寻找CPU，无限死循环
> 
> 求教大雕的其它版本如何编译啊？

大雕源码何来的其他版本？