281677160/build-actions #141: 大佬帮忙看看什么问题

authon, 2024-02-29:
make[3] -C package/libs/libusb compile
 make[3] -C feeds/packages/libs/gperf host-compile
 make[3] -C feeds/packages/utils/v2dat compile
 make[3] -C feeds/passwall3/brook compile
    ERROR: package/feeds/packages/v2dat failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/Ledemaster/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/Ledemaster/openwrt/scripts/config'
make[1]: Entering directory '/Ledemaster/openwrt'
make[2]: Entering directory '/Ledemaster/openwrt'

281677160, 2024-03-03:
问题已修复

drw2005, 2024-03-04:
正常了。

closed 2024-03-05 by 281677160