281677160/build-actions #142: 2.25更新的小版本，提示太大编译失败，之前同样的编译内容能顺利通过，请问如何解决？

xiaofanxinxiang1, 2024-03-01:
1.编译ssh中增加的插件
    已选插件列表
        1、luci-app-accesscontrol
        2、luci-app-adguardhome
        3、luci-app-argon-config
        4、luci-app-arpbind
        5、luci-app-autoreboot
        6、luci-app-autoupdate
        7、luci-app-ddns
        8、luci-app-fileassistant
        9、luci-app-filebrowser
        10、luci-app-filetransfer
        11、luci-app-firewall
        12、luci-app-frpc
        13、luci-app-frps
        14、luci-app-ipsec-server
        15、luci-app-mosdns
        16、luci-app-netdata
        17、luci-app-nlbwmon
        18、luci-app-ntpc
        19、luci-app-openclash
        20、luci-app-passwall2
        21、luci-app-poweroff
        22、luci-app-pptp-server
        23、luci-app-samba4
        24、luci-app-smartdns
        25、luci-app-socat
        26、luci-app-ssr-plus
        27、luci-app-ttyd
        28、luci-app-upnp
        29、luci-app-wireguard
        30、luci-app-zerotier
        31、luci-theme-Butterfly
        32、luci-theme-Light
        33、luci-theme-argon
        34、luci-theme-argon-dark-mod
        35、luci-theme-darkmatter
        36、luci-theme-design
        37、luci-theme-ifit
        38、luci-theme-kucat
        39、luci-theme-material
2.该固件编译过程中提示的问题
  g++ -o /Ledemaster/openwrt/build_dir/hostpkg/node-v16.19.1/out/Release/obj.target/icutools/deps/icu-small/source/common/ucmndata.o ../deps/icu-small/source/common/ucmndata.cpp '-DV8_DEPRECATION_WARNINGS' '-DV8_IMMINENT_DEPRECATION_WARNINGS' '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DU_COMMON_IMPLEMENTATION=1' '-DU_I18N_IMPLEMENTATION=1' '-DU_IO_IMPLEMENTATION=1' '-DU_TOOLUTIL_IMPLEMENTATION=1' '-DU_ATTRIBUTE_DEPRECATED=' '-D_CRT_SECURE_NO_DEPRECATE=' '-DU_STATIC_IMPLEMENTATION=1' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' -I../deps/icu-small/source/common -I../deps/icu-small/source/i18n -I../deps/icu-small/source/tools/toolutil  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -Wno-deprecated-declarations -Wno-strict-aliasing -O3 -fno-omit-frame-pointer -fno-exceptions -std=gnu++14 -frtti -MMD -MF /Ledemaster/openwrt/build_dir/hostpkg/node-v16.19.1/out/Release/.deps//Ledemaster/openwrt/build_dir/hostpkg/node-v16.19.

> ### **This step has been truncated due to its large size. Download the full logs from the  menu once the workflow run has completed.**
请问应该如何调整设置才能编译成功

281677160, 2024-03-02:
这点日志看啥？或者是上游源码问题吧，再次编译试试吧

xiaofanxinxiang1, 2024-03-03:
> 这点日志看啥？或者是上游源码问题吧，再次编译试试吧

重新编译了十几次都是失败，然后看了一下日志，在200行左右会出现如下提示
“ make[3] -C package/base-files compile
    ERROR: package/feeds/packages/node [host] failed to build.
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 1
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make[2]: Entering directory '/Ledemaster/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/Ledemaster/openwrt/scripts/config'
make[1]: Entering directory '/Ledemaster/openwrt'
make[2]: Entering directory '/Ledemaster/openwrt'
make[3]: Entering directory '/Ledemaster/openwrt/target/linux'
make[4]: Entering directory '/Ledemaster/openwrt/target/linux/x86'”

但是不会停止编译，会继续到2万8千行的时候出现上面的问题，不知道是什么原因。感谢您的付出和分析
谢谢

xiaofanxinxiang1, 2024-03-03:
> 这点日志看啥？或者是上游源码问题吧，再次编译试试吧

在最后会出现报错信息如下
  g++ -o /Ledemaster/openwrt/build_dir/hostpkg/node-v16.19.1/out/Release/obj.target/v8_libbase/deps/v8/src/base/cpu.o ../deps/v8/src/base/cpu.cc '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DV8_TARGET_ARCH_X64' '-DV8_HAVE_TARGET_OS' '-DV8_TARGET_OS_LINUX' '-DV8_EMBEDDER_STRING="-node.24"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_INTL_SUPPORT' '-DV8_ATOMIC_OBJECT_FIELD_WRITES' '-DV8_ATOMIC_MARKING_STATE' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DV8_ENABLE_WEBASSEMBLY' '-DV8_ALLOCATION_FOLDING' '-DV8_ALLOCATION_SITE_TRACKING' '-DV8_ADVANCED_BIGINT_ALGORITHMS' '-DBUILDING_V8_BASE_SHARED' -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++14 -MMD -MF /Ledemaster/openwrt/build_dir/hostpkg/node-v16.19.1/out/Release/.deps//Ledemaster/openwrt/build_dir/hostpkg/node-v16.19.1/out/Release/obj.target/v8_libbase/deps/v8/src/base/cpu.o.d.raw -I/Ledemaster/openwrt/staging_dir/host/include -I/Ledemaster/openwrt/staging_dir/hostpkg/include -I/Ledemaster/openwrt/staging_dir/target-x86_64_musl/host/include  -c
In file included from ../deps/v8/src/base/macros.h:12,
                 from ../deps/v8/src/base/cpu.h:17,
                 from ../deps/v8/src/base/cpu.cc:5:
../deps/v8/src/base/logging.h: In function 'auto v8::base::detail::GetUnderlyingEnumTypeForPrinting(T)':
../deps/v8/src/base/logging.h:154:26: error: 'uint16_t' was not declared in this scope
  154 |                          uint16_t> >;
      |                          ^~~~~~~~
../deps/v8/src/base/logging.h:17:1: note: 'uint16_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   16 | #include "src/base/template-utils.h"
  +++ |+#include <cstdint>
   17 | 
../deps/v8/src/base/logging.h:154:34: error: template argument 3 is invalid
  154 |                          uint16_t> >;
      |                                  ^
../deps/v8/src/base/logging.h:154:36: error: template argument 3 is invalid
  154 |                          uint16_t> >;
      |                                    ^
../deps/v8/src/base/logging.h:154:37: error: expected identifier before ';' token
  154 |                          uint16_t> >;
      |                                     ^
../deps/v8/src/base/logging.h:155:22: error: 'int_t' does not name a type; did you mean 'wint_t'?
  155 |   return static_cast<int_t>(static_cast<underlying_t>(val));
      |                      ^~~~~
      |                      wint_t
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:314:8: error: 'uint64_t' does not name a type
  314 | inline uint64_t make_uint64(uint32_t high, uint32_t low) {
      |        ^~~~~~~~
../deps/v8/src/base/macros.h:14:1: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   13 | #include "src/base/platform/wrappers.h"
  +++ |+#include <cstdint>
   14 | 
../deps/v8/src/base/macros.h:320:25: error: 'intptr_t' has not been declared
  320 | inline T RoundDown(T x, intptr_t m) {
      |                         ^~~~~~~~
../deps/v8/src/base/macros.h:326:11: error: 'intptr_t' has not been declared
  326 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundDown(T)':
../deps/v8/src/base/macros.h:330:17: error: 'm' was not declared in this scope
  330 |   STATIC_ASSERT(m != 0 && ((m & (m - 1)) == 0));
      |                 ^
../deps/v8/src/base/macros.h:182:43: note: in definition of macro 'STATIC_ASSERT'
  182 | #define STATIC_ASSERT(test) static_assert(test, #test)
      |                                           ^~~~
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:336:23: error: 'intptr_t' has not been declared
  336 | inline T RoundUp(T x, intptr_t m) {
      |                       ^~~~~~~~
../deps/v8/src/base/macros.h:340:11: error: 'intptr_t' has not been declared
  340 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundUp(T)':
../deps/v8/src/base/macros.h:343:20: error: 'm' was not declared in this scope
  343 |   return RoundDown<m, T>(static_cast<T>(x + (m - 1)));
      |                    ^
../deps/v8/src/base/macros.h: In function 'void* AlignedAddress(void*, size_t)':
../deps/v8/src/base/macros.h:354:51: error: 'uintptr_t' does not name a type
  354 |   return reinterpret_cast<void*>(reinterpret_cast<uintptr_t>(address) &
      |                                                   ^~~~~~~~~
../deps/v8/src/base/macros.h:354:51: note: 'uintptr_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
../deps/v8/src/base/macros.h:355:47: error: 'uintptr_t' does not name a type
  355 |                                  ~static_cast<uintptr_t>(alignment - 1));
      |                                               ^~~~~~~~~
[... cut ...]
time: package/feeds/packages/node/host-compile#351.01#38.69#389.25
    ERROR: package/feeds/packages/node [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/node/host/compile] Error 1
make[2]: Leaving directory '/Ledemaster/openwrt'
make[1]: *** [package/Makefile:110: /Ledemaster/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Ledemaster/openwrt'
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

感谢帮忙分析下是何原因谢谢

281677160, 2024-03-03:
应该是node-v16.19.1这个版本太低了，用高版本替换就编译通过了，你自己去测试一下吧

xiaofanxinxiang1, 2024-03-04:
> 应该是node-v16.19.1这个版本太低了，用高版本替换就编译通过了，你自己去测试一下吧

不好意思，我不是太专业，请问应该如何替换这个node？麻烦您拨冗解答，谢谢

closed 2024-03-05 by 281677160