281677160/build-actions #144: 希望增加hanwckf/immortalwrt-mt798x源码编译

ShmilyStar, 2024-03-12:
请问能否增加hanwckf/immortalwrt-mt798x源码编译命令，自己尝试修改，总是出问题

281677160, 2024-03-12:
跟源作者immortalwrt的21.02有啥区别？

ShmilyStar, 2024-03-12:
> 跟源作者immortalwrt的21.02有啥区别？
针对mt798x的闭源驱动，比开源驱动更稳定，市面上现在的大部分路由器都是mt798x处理器的，具体项目简介在这里https://cmi.hanwckf.top/p/immortalwrt-mt798x/

281677160, 2024-04-04:
> > 跟源作者immortalwrt的21.02有啥区别？
> > 针对mt798x的闭源驱动，比开源驱动更稳定，市面上现在的大部分路由器都是mt798x处理器的，具体项目简介在这里https://cmi.hanwckf.top/p/immortalwrt-mt798x/
[image]

ShmilyStar, 2024-04-22:
> > > 跟源作者immortalwrt的21.02有啥区别？
> > > 针对mt798x的闭源驱动，比开源驱动更稳定，市面上现在的大部分路由器都是mt798x处理器的，具体项目简介在这里https://cmi.hanwckf.top/p/immortalwrt-mt798x/
> > > ![QQ截图20240404132926](https://private-user-images.githubusercontent.com/66651997/319455224-07d493eb-ca0b-4acf-82c1-6699c0e0c82f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM3NTg1MjcsIm5iZiI6MTcxMzc1ODIyNywicGF0aCI6Ii82NjY1MTk5Ny8zMTk0NTUyMjQtMDdkNDkzZWItY2EwYi00YWNmLTgyYzEtNjY5OWMwZTBjODJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDIyVDAzNTcwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRmNjFhMjVjN2JkZGNhM2Q1MTYzOThkMWNjN2NkN2E5ZDc4ODExNTJkYzdkZjVjOTNjM2VjN2VlOGM2ZmMzNWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XKw9YtVj9hzD8vzHP31_qPl5k7BN0xdh8Lr0cktg_mY)

感谢，经测试，可以拉取这个仓库的源码编译，但是里面的luci-lucky插件不知道为啥一直编译不上去

ShmilyStar, 2024-04-22:
是显示编译上去了，有lucky界面，软件包里面也有lucky的三个软件包，但是lucky不起作用，不能启动

closed 2025-03-22 by 281677160