281677160/build-actions #165: 编译immortalwrt时找不到homeproxy插件。。。

drw2005, 2024-08-23:
大佬能否把https://github.com/immortalwrt/homeproxy插件加进来？
谢谢！

zh15933, 2024-08-23:
有啊 我编译的时候都看到了 我自己拉取了才发现 源码自带

zh15933, 2024-08-23:
! [Snipaste_2024-08-23_23-55-19] (https://github.com/user-attachments/assets/fee7f848-9766-405e-911a-c8c59d752ba3)

我进去看了一眼 确实是  没看错

zh15933, 2024-08-23:
我进去看了一眼 确实是 没看错

[image]

drw2005, 2024-08-24:
> 我进去看了一眼 确实是 没看错
> 
> ![Snipaste_2024-08-23_23-55-19](https://private-user-images.githubusercontent.com/32957590/360989455-6251fdde-7919-42ad-ba96-63f9131e8ee6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ0NTg0NjQsIm5iZiI6MTcyNDQ1ODE2NCwicGF0aCI6Ii8zMjk1NzU5MC8zNjA5ODk0NTUtNjI1MWZkZGUtNzkxOS00MmFkLWJhOTYtNjNmOTEzMWU4ZWU2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI0VDAwMDkyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUxODU5MTM3ZDVjODA5YzY2NzIxYTYxZDNlNmM5ZDgwNzBjYTY5NDk5ZTk5ZTQyZTFkOGIwY2M3NDEyYjAzN2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.O-pbrBYQX37j3ARQWkT9zsXIDTgufbm8XEbG3PLXXbQ)

那可能我没看到，谢谢指点。

drw2005, 2024-08-26:
今天仔细看了一下，确实没有。
[image]

closed 2024-08-27 by drw2005