281677160/build-actions #172: LEDE源码能否加入23.05分支？

drw2005, 2024-09-24:
大佬好久没出现了，有时间把openwrt-23.05加入进来吧。

281677160, 2025-03-08:
弄好了，现在是23.05的LUCI，你是本地编译的？

drw2005, 2025-03-09:
> 弄好了，现在是23.05的LUCI，你是本地编译的？

对，本地编译。

好久没出现了，以为你放弃了。

281677160, 2025-03-09:
> > 弄好了，现在是23.05的LUCI，你是本地编译的？
> 
> 对，本地编译。
> 
> 好久没出现了，以为你放弃了。

我是放弃了啊，我好久没用openwrt了，就手机安装个APP使用，我手机升级鸿蒙5.0系统后，没办法安装翻墙APP，我才用openwrt的，自己想编译都搞不成，就顺便都搞搞咯，本地编译还能用吗？

drw2005, 2025-03-10:
> > > 弄好了，现在是23.05的LUCI，你是本地编译的？
> > 
> > 
> > 对，本地编译。
> > 好久没出现了，以为你放弃了。
> 
> 我是放弃了啊，我好久没用openwrt了，就手机安装个APP使用，我手机升级鸿蒙5.0系统后，没办法安装翻墙APP，我才用openwrt的，自己想编译都搞不成，就顺便都搞搞咯，本地编译还能用吗？

别放弃啊，没事过来更新一下就行，你这个最好用方便。
你这次更新后，本地编译我还没试过，准备今天试试。
另外，插件库也更新一下吧。

drw2005, 2025-03-20:
> > > > > 弄好了，现在是23.05的LUCI，你是本地编译的？
> > > > 
> > > > 
> > > > 对，本地编译。
> > > > 好久没出现了，以为你放弃了。
> > > 
> > > 
> > > 我是放弃了啊，我好久没用openwrt了，就手机安装个APP使用，我手机升级鸿蒙5.0系统后，没办法安装翻墙APP，我才用openwrt的，自己想编译都搞不成，就顺便都搞搞咯，本地编译还能用吗？
> > 
> > 
> > 别放弃啊，没事过来更新一下就行，你这个最好用方便。 你这次更新后，本地编译我还没试过，准备今天试试。 另外，插件库也更新一下吧。
> 
> 插件？更新啥？

passwall2，看到已经同步了。