281677160/build-actions #177: openclash内核选择问题

authon, 2024-10-23:
export OpenClash_branch="0" 
export OpenClash_Core="2"

你好，请问这里怎么设置只下载meta内核，内核那里选择2貌似刷进去的时候没有下载好内核，要重新点击下载才会下载内核

authon, 2024-10-24:
在diy里面加这个会编译失败
# ##------------- meta core ---------------------------------
# curl -sL -m 30 --retry 2 https://raw.githubusercontent.com/vernesong/OpenClash/core/master/meta/clash-linux-arm64.tar.gz -o /tmp/clash.tar.gz
# tar zxvf /tmp/clash.tar.gz -C /tmp >/dev/null 2>&1
# chmod +x /tmp/clash >/dev/null 2>&1
# mv /tmp/clash feeds/luci/applications/luci-app-openclash/root/etc/openclash/core/clash_meta >/dev/null 2>&1
# ##---------------------------------------------------

281677160, 2025-03-08:
现在不需要添加内核了，插件增加了代理下载内核的功能，编译好后，自己下载一下就行

authon, 2025-03-08:
> 现在不需要添加内核了，插件增加了代理下载内核的功能，编译好后，自己下载一下就行

好的好的

closed 2025-03-22 by 281677160