281677160/build-actions #191: 生成cmcc-rax3000me 固件报错

tom4733, 2025-03-31:
生成cmcc-rax3000me 固件报错  一键 去生成 .config 跑完代码 保存了   cmcc-rax3000me 也出不来 一直是360T7的.config ,不知道是啥问题 能帮忙生成cmcc-rax3000me 固件 带ssr 和 passwall 插件的吗谢谢

281677160, 2025-03-31:
什么源码？SSH进去看过有没有cmcc-rax3000me这个机型了？

tom4733, 2025-03-31:
> 什么源码？SSH进去看过有没有cmcc-rax3000me这个机型了？
[Mt798x.yml](https://github.com/281677160/build-actions/blob/main/.github/workflows/Mt798x.yml)  这个有的

281677160, 2025-03-31:
[image]

这不是好好的吗？配置文件不是可以生成了？360T7的文件进去，然后改选机型就出来了，那么难吗？如果是编译后的固件有问题，那就另说了，我的说明都写了，21.02跟23.05的【mt7981和mt7986】都是同步2410的机型文件，同步是同步了，就没了解过编译出的固件会不会有问题，因为我也没机子测试，还有就是同时选择多个插件，要注意固件会不会超过固件固定的大小而编译不出来固件

tom4733, 2025-03-31:
> [image]
> 
> 这不是好好的吗？配置文件不是可以生成了？360T7的文件进去，然后改选机型就出来了，那么难吗？如果是编译后的固件有问题，那就另说了，我的说明都写了，21.02跟23.05的【mt7981和mt7986】都是同步2410的机型文件，同步是同步了，就没了解过编译出的固件会不会有问题，因为我也没机子测试，还有就是同时选择多个插件，要注意固件会不会超过固件固定的大小而编译不出来固件

怪事我本地一键 去生成 .config 一直跳出来都是 360T7的  晕死

tom4733, 2025-03-31:
> [image]
> 
> 这不是好好的吗？配置文件不是可以生成了？360T7的文件进去，然后改选机型就出来了，那么难吗？如果是编译后的固件有问题，那就另说了，我的说明都写了，21.02跟23.05的【mt7981和mt7986】都是同步2410的机型文件，同步是同步了，就没了解过编译出的固件会不会有问题，因为我也没机子测试，还有就是同时选择多个插件，要注意固件会不会超过固件固定的大小而编译不出来固件

等你那跑完了   我下载试试

281677160, 2025-03-31:
> > [image]
> > 这不是好好的吗？配置文件不是可以生成了？360T7的文件进去，然后改选机型就出来了，那么难吗？如果是编译后的固件有问题，那就另说了，我的说明都写了，21.02跟23.05的【mt7981和mt7986】都是同步2410的机型文件，同步是同步了，就没了解过编译出的固件会不会有问题，因为我也没机子测试，还有就是同时选择多个插件，要注意固件会不会超过固件固定的大小而编译不出来固件
> 
> [image] 这是啥问题啊？ 只有3个lan wan 和 wan6 没

不知道，这些东西不是自己建立的吗？你用24.10的固件试试看，如果都是一样问题，你就去源码作者那里问吧，如果24.10没问题，就告诉我，我看看是不是我拉取机型文件的时候少了什么

tom4733, 2025-03-31:
> 不知道，这些东西不是自己建立的吗？你用24.10的固件试试看，如果都是一样问题，你就去源码作者那里问吧，如果24.10没问题，就告诉我，我看看是不是我拉取机型文件的时候少了什么

24.10 和23.05 刷不进去 只有这个 21.0 可以刷  就是只要3个lan

tom4733, 2025-03-31:
> > > > > [image]这不是好好的吗？配置文件不是可以生成了？360T7的文件进去，然后改选机型就出来了，那么难吗？如果是编译后的固件有问题，那就另说了，我的说明都写了，21.02跟23.05的【mt7981和mt7986】都是同步2410的机型文件，同步是同步了，就没了解过编译出的固件会不会有问题，因为我也没机子测试，还有就是同时选择多个插件，要注意固件会不会超过固件固定的大小而编译不出来固件
> > > > 
> > > > 
> > > > [image]这是啥问题啊？ 只有3个lan wan 和 wan6 没
> > > 
> > > 
> > > 不知道，这些东西不是自己建立的吗？你用24.10的固件试试看，如果都是一样问题，你就去源码作者那里问吧，如果24.10没问题，就告诉我，我看看是不是我拉取机型文件的时候少了什么
> > 
> > 
> > 24.10 和23.05 刷不进去 只有这个 21.0 可以刷 就是只要3个lan
> 
> 机型文件重新整理了一下，再编译了，到时候看看吧，还不行，我也没办法了，24.10的内核跟机型那些，我看他那些源码基本都通用，没道理你的机子刷不了24.10固件啊

不行 24.10 刷了没反应  只有21.02 可以刷  

[image]

281677160, 2025-03-31:
> 不行 24.10 刷了没反应 只有21.02 可以刷
>
> [image]

https://github.com/281677160/autobuild/releases/download/Update-mediatek/21.02-Mt798x-cmcc_rax3000me-1743429590-sysupgrade-149ac3.bin

这个刚编译好的，去试试看什么情况

tom4733, 2025-03-31:
> https://github.com/281677160/autobuild/releases/download/Update-mediatek/21.02-Mt798x-cmcc_rax3000me-1743429590-sysupgrade-149ac3.bin
>
> 这个刚编译好的，去试试看什么情况

好的 谢谢 我试试

281677160, 2025-03-31:
> > > > > > > > > [image]这不是好好的吗？配置文件不是可以生成了？360T7的文件进去，然后改选机型就出来了，那么难吗？如果是编译后的固件有问题，那就另说了，我的说明都写了，21.02跟23.05的【mt7981和mt7986】都是同步2410的机型文件，同步是同步了，就没了解过编译出的固件会不会有问题，因为我也没机子测试，还有就是同时选择多个插件，要注意固件会不会超过固件固定的大小而编译不出来固件
> > > > > > > > 
> > > > > > > > 
> > > > > > > > [image]这是啥问题啊？ 只有3个lan wan 和 wan6 没
> > > > > > > 
> > > > > > > 
> > > > > > > 不知道，这些东西不是自己建立的吗？你用24.10的固件试试看，如果都是一样问题，你就去源码作者那里问吧，如果24.10没问题，就告诉我，我看看是不是我拉取机型文件的时候少了什么
> > > > > > 
> > > > > > 
> > > > > > 24.10 和23.05 刷不进去 只有这个 21.0 可以刷 就是只要3个lan
> > > > > 
> > > > > 
> > > > > 机型文件重新整理了一下，再编译了，到时候看看吧，还不行，我也没办法了，24.10的内核跟机型那些，我看他那些源码基本都通用，没道理你的机子刷不了24.10固件啊
> > > > 
> > > > 
> > > > 不行 24.10 刷了没反应 只有21.02 可以刷
> > > > [image]
> > > 
> > > 
> > > https://github.com/281677160/autobuild/releases/download/Update-mediatek/21.02-Mt798x-cmcc_rax3000me-1743429590-sysupgrade-149ac3.bin
> > > 这个刚编译好的，去试试看什么情况
> > 
> > 
> > 好的 谢谢 我试试
> 
> [image]
> 
> 可以了 出来了

那就行了，23.05应该也快编译好了，一会好了，再去看看23.05能不能用吧

tom4733, 2025-03-31:
> 那就行了，23.05应该也快编译好了，一会好了，再去看看23.05能不能用吧

好的嗯嗯 ，大佬 .config 怎么编写的啊？一键 去生成 .config 跑完代码 保存了 cmcc-rax3000me 也出不来 一直是360T7的.config 

[image]

281677160, 2025-03-31:
那么长时间的东西，鬼记得那么多，我也是服了，你复制一个360T7的文件，把文件名字改成你的机型名字，编译的时候就用这个文件来编译，然后进SSH改一下机型不就好了，那么简单的事情

281677160, 2025-03-31:
https://github.com/281677160/autobuild/releases/download/Update-mediatek/23.05-Mt798x-cmcc_rax3000me-1743430461-sysupgrade-6136a7.bin

这个是23.05的

closed 2025-04-01 by 281677160