281677160/build-actions #194: 昨天编译都正常 ，今天怎么跑都报这样 是啥问题啊

tom4733, 2025-04-05:
[image]昨天编译都正常 ，今天怎么跑都报这样 是啥问题啊

281677160, 2025-04-05:
不要开缓存看看

tom4733, 2025-04-05:
> 不要开缓存看看

[image] 现在是这样了

tom4733, 2025-04-05:
> > 不要开缓存看看
> 
> [image] 现在是这样了

[image] 没缓存呀

281677160, 2025-04-05:
你用的是我测试仓库，有什么问题，自己解决

tom4733, 2025-04-05:
> > 不要开缓存看看
> 
> [image] 现在是这样了

没缓存啊

tom4733, 2025-04-05:
> 你用的是我测试仓库，有什么问题，自己解决

哪个不是测试的啊

tom4733, 2025-04-05:
> > 你用的是我测试仓库，有什么问题，自己解决
> 
> 哪个不是测试的啊

[image]这是是测试的吗

closed 2025-04-05 by tom4733