281677160/build-actions #201: x86 编译错误

harry987654, 2025-04-20:
Building targets
github.com/winfsp/cgofuse/fuse
# github.com/winfsp/cgofuse/fuse
../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.1-0.20230130140708-f87f5db493b5/fuse/host_cgo.go:117:10: fatal error: fuse.h: No such file or directory
  117 | #include <fuse.h>
      |          ^~~~~~~~
compilation terminated.

make[3]: *** [Makefile:99: /Lede2305/openwrt/build_dir/target-x86_64_musl/alist-3.44.0/.built] Error 1
make[3]: Leaving directory '/Lede2305/openwrt/feeds/danshui/luci-app-alist/alist'
time: package/feeds/danshui/alist/compile#8.21#2.93#8.75
    ERROR: package/feeds/danshui/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/danshui/alist/compile] Error 1
make[2]: Leaving directory '/Lede2305/openwrt'
make[1]: *** [package/Makefile:110: /Lede2305/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/Lede2305/openwrt'
make: *** [/Lede2305/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

281677160, 2025-04-20:
你重新编译试试看

281677160, 2025-04-21:
兄弟什么情况了？能不能编译成功了？

harry987654, 2025-04-22:
又编译了一次，还是出现错误。
 make[2] package/install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/Lede2305/openwrt/include/toplevel.mk:231: world] Error 1
config.guess: cannot create a temporary directory in /tmp
config.guess: cannot create a temporary directory in /tmp
config.guess: cannot create a temporary directory in /tmp
awk: (FILENAME=- FNR=2) warning: error writing standard output: No space left on device
awk: cmd. line:14: (FILENAME=- FNR=60) fatal: print to "standard output" failed: No space left on device
Error: Unable to process file command 'step_summary' successfully.
Error: No space left on device : '/home/runner/runners/2.323.0/_diag/Worker_20250422-004929-utc.log'
Error: No space left on device : '/home/runner/runners/2.323.0/_diag/pages/6bcb45b6-1faf-4350-8a9b-8cc432b2b325_ba48a14c-dcc2-5fe0-a16b-5ac5699e6038_1.log'

281677160, 2025-04-22:
> 又编译了一次，还是出现错误。 make[2] package/install make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on make: *** [...]

你都编译了啥啊？这个是已经要成功编译了，没空间了，看到make[2] package/install这个没有？到了这里就成功了，在弄插件了，没空间了

closed 2025-04-24 by 281677160