281677160/build-actions #203: 大佬，麻烦请问一下编译版本该怎么选择

shengonglin, 2025-04-23:
大佬，有一个问题。从去年开始用你的源码编译了很多次了。因为是新手，一开始都是跟着说明编译的，今天发现一直以来编译的都是snapshot版本，如果想编译immortalwrt的x86-64的24.10.1稳定版本的话。有什么方法么？比如在配置文件的哪里修改就可以了。因为现在按照教程就只知道选择大版本号，其他的就不知道该怎么定义了。

281677160, 2025-04-23:
源码拉取的是什么就是什么

[image]

shengonglin, 2025-04-23:
好的。谢谢大佬解答。超级感谢

shengonglin, 2025-04-23:
大佬，实在不好意思。问了deepseek，最后还是没搞成，能否告知一下该如何加代码，我只想编译immortalwrt的x86-64的24.10.1的版本，但github里他的分支没有这个版本，tags里面有。

281677160, 2025-04-23:
````
    - name: 修改版本号
      run: |
        cd openwrt
        git tag -l
        git describe --tags
        git checkout v24.10.1
````

[image]

shengonglin, 2025-04-23:
感谢大佬帮忙，超级感谢。去测试了一下可以。大佬真牛！

closed 2025-04-23 by shengonglin