281677160/build-actions #205: 大佬,imm23.05编译出来好像是原生openwrt

zh15933, 2025-04-23:
大佬,imm23.05编译出来好像是原生openwrt

[image]

编译完成后刷入  确定是原生openwrt              

24.10 看起来是正常的

zh15933, 2025-04-23:
这个是imm 2410的界面

[image]

但是在选择插件的时候,发现openclash没有了?我需要在diy文件里,把opencalsh相关的选项改成1或者2?

[image]

281677160, 2025-04-23:
官方跟天灵的23.05，内核版本升级到180后，不知道为什么死活编译不成功，只能回退到没修改版本之前的版本，也就只回退到2周前的，不会给它回退到官方的去那么夸张吧
openclash看diy扩展的说明啊，现在是选择1或者2了，选了后，不管你在这里勾选不勾选，都会自动默认给你选择把这个插件编译进去了

zh15933, 2025-04-23:
> 官方跟天灵的23.05，内核版本升级到180后，不知道为什么死活编译不成功，只能回退到没修改版本之前的版本，也就只回退到2周前的，不会给它回退到官方的去那么夸张吧 openclash看diy扩展的说明啊，现在是选择1或者2了，选了后，不管你在这里勾选不勾选，都会自动默认给你选择把这个插件编译进去了

emmmm imm2305  我编译出来确实是openwrt, 因为在Dnsmasq界面那里,没有DNS重定向这个功能

281677160, 2025-04-23:
我也头疼，不管了，我看了源码，几小时之前还有修改，不知道修复了没，我把回退操作取消了，看看能编译不。反正官方的跟天灵都一样，内核升级到180后死活编译不了

281677160, 2025-04-24:
上游已经修复了180版本号编译不了的问题了

closed 2025-04-24 by zh15933