281677160/build-actions #209: 总是出现空间不足的错误 System.IO.IOException: No space left on device

harry987654, 2025-04-27:
编译前加了个磁盘命令，发现只剩19G可用了，是不是导致空间不足的问题？那么如何增加存储空间呢？

Run df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        73G   54G   19G  75% /
tmpfs           7.9G  172K  7.9G   1% /dev/shm
tmpfs           3.2G  1.1M  3.2G   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
/dev/sda15      105M  6.1M   99M   6% /boot/efi
/dev/sdb1        74G  4.1G   66G   6% /mnt
tmpfs           1.6G   12K  1.6G   1% /run/user/1001

281677160, 2025-04-27:
本身不是有增加空间的？

[image]

harry987654, 2025-05-03:
逐步增加插件，终于编译成功了，看样子不能一次勾选太多的插件啊。
另外想增加一个主题插件，不知道把代码放到哪里？
# luci-theme-ifit
src-git ifit https://github.com/kenzok8/luci-theme-ifit.git

281677160, 2025-05-03:
> 逐步增加插件，终于编译成功了，看样子不能一次勾选太多的插件啊。 另外想增加一个主题插件，不知道把代码放到哪里？
> 
> # luci-theme-ifit
> src-git ifit https://github.com/kenzok8/luci-theme-ifit.git

这个主题是18.06的，其他都不兼容

closed 2025-05-10 by 281677160