281677160/build-actions #210: 显示触发编译成功，但是没有后续工作流

TuringTrail, 2025-04-29:
[image]
显示触发编译Mt798x-24.10-xiaomi_redmi-router-ax6000成功 ，但是没有后续工作流

281677160, 2025-04-29:
看了你的仓库，没有触发编译，不知道什么原因显示成功的，等我有空了看看

281677160, 2025-04-29:
测试了一下，触发是没问题啊，你把密匙重新制作了试试

TuringTrail, 2025-04-29:
> 测试了一下，触发是没问题啊，你把密匙重新制作了试试

我重新创建了分支编译的是否跟这个有关呢？

TuringTrail, 2025-04-29:
> 测试了一下，触发是没问题啊，你把密匙重新制作了试试

我是用fork主分支触发是没问题的，如果使用新分支则不能触发

281677160, 2025-04-29:
> > 测试了一下，触发是没问题啊，你把密匙重新制作了试试
> 
> 我是用fork主分支触发是没问题的，如果使用新分支则不能触发

要用主分支才行，你改变分支名称的话，就把那个分支设置成主分支，就是你打开仓库就看到的分支

TuringTrail, 2025-04-29:
> > > 测试了一下，触发是没问题啊，你把密匙重新制作了试试
> > 
> > 
> > 我是用fork主分支触发是没问题的，如果使用新分支则不能触发
> 
> 要用主分支才行，你改变分支名称的话，就把那个分支设置成主分支，就是你打开仓库就看到的分支

能否增加一个选项支持多分支编译呢

TuringTrail, 2025-04-29:
我看是在compile.yml 固定设置了编译主分支

281677160, 2025-04-29:
> 我看是在compile.yml 固定设置了编译主分支

要多改一个分支号，很烦的，有时间再说吧

TuringTrail, 2025-04-29:
> > 我看是在compile.yml 固定设置了编译主分支
> 
> 你现在测试一下看看能不能用其他分支触发

正常了 感谢

closed 2025-04-29 by TuringTrail