281677160/build-actions #214: 大佬 又遇到问题了

shengonglin, 2025-05-11:
今天用immortalwrt的master源码编译了两次，一次是默认内核，一次是替换的6.12的内核。但是编译完成以后，打开登录界面就报错，配置全是默认的，什么都没改动，只是把ip地址改成了192.168.100.10然后进行的编译。

大佬能帮忙看看是什么原因么。

报错截图：

[image]

shengonglin, 2025-05-12:
大佬 问了ai，说是少了个openclash的datatypes.lua的文件。然后看了一下确实和之前编译的对比，少了这个文件。之前编译的在这两个文件夹都有。

[image]

而现在编译master版本的是cbi文件夹和datatypes.lua都没有，不知道这是不是openclash在最新的master版本下的问题。然后把之前编译的24.10之类的版本里面的这个文件复制过master版本后就都正常了。但因为rom文件夹也没有。所以万一使用还原的话就又出问题了。大佬如果有时间就看看是源码本身的问题还是openclash的问题。可以的话是不是能修复之类的。辛苦了~~

PS.增加文件后能正常进入网页了，但是点openclash，又出现新的错误，还是缺少文件。

[image]

总结就是最新immortalwrt的master版本。编译openclash有大问题。但是大佬的脚本默认会编译openclash。。。。

281677160, 2025-05-12:
修复好了。你重新编译测试一下

shengonglin, 2025-05-12:
好的。感谢大佬~~每次都好迅速的！

closed 2025-05-12 by shengonglin