281677160/build-actions #215: 好像又不能编译了

AlieZ777, 2025-05-17:
2025-05-17T10:15:40.0101668Z ##[group]Run 281677160/common@need
2025-05-17T10:15:40.0101928Z env:
2025-05-17T10:15:40.0102114Z   GIT_REPOSITORY: AlieZ777/build-actions
2025-05-17T10:15:40.0102517Z   REPO_TOKEN: ***
2025-05-17T10:15:40.0102697Z   PUSH_PLUS_TOKEN: 
2025-05-17T10:15:40.0102895Z   TELEGRAM_BOT_TOKEN: 
2025-05-17T10:15:40.0103071Z   TELEGRAM_CHAT_ID: 
2025-05-17T10:15:40.0103245Z   TZ: Asia/Shanghai
2025-05-17T10:15:40.0103412Z   FOLDER_NAME: Lienol
2025-05-17T10:15:40.0103583Z   BENDI_VERSION: 2
2025-05-17T10:15:40.0103749Z   SOURCE_CODE: LIENOL
2025-05-17T10:15:40.0103920Z   REPO_BRANCH: 24.10
2025-05-17T10:15:40.0104109Z   INFORMATION_NOTICE: false
2025-05-17T10:15:40.0104501Z   RETAIN_MINUTE: 10080
2025-05-17T10:15:40.0104676Z   KEEP_LATEST: 30
2025-05-17T10:15:40.0104843Z   SSH_ACTION: true
2025-05-17T10:15:40.0105009Z   UPLOAD_FIRMWARE: false
2025-05-17T10:15:40.0105200Z   UPLOAD_RELEASE: true
2025-05-17T10:15:40.0105385Z   CACHEWRTBUILD_SWITCH: true
2025-05-17T10:15:40.0105587Z   UPDATE_FIRMWARE_ONLINE: true
2025-05-17T10:15:40.0105891Z   GITHUB_LINK: https://github.com/AlieZ777/build-actions
2025-05-17T10:15:40.0106167Z   GIT_ACTOR: AlieZ777
2025-05-17T10:15:40.0106352Z   WAREHOUSE_MAN: build-actions
2025-05-17T10:15:40.0106541Z   RUN_NUMBER: 20
2025-05-17T10:15:40.0106707Z   GIT_REFNAME: main
2025-05-17T10:15:40.0106879Z   COMPILATION_INFORMATION: true
2025-05-17T10:15:40.0107084Z   CONFIG_FILE: x86_64
2025-05-17T10:15:40.0107345Z   HOME_PATH: /home/runner/work/build-actions/build-actions/openwrt
2025-05-17T10:15:40.0107738Z   OPERATES_PATH: /home/runner/work/build-actions/build-actions/operates
2025-05-17T10:15:40.0108172Z   COMPILE_PATH: /home/runner/work/build-actions/build-actions/operates/Lienol
2025-05-17T10:15:40.0108794Z   BUILD_DIY: /home/runner/work/build-actions/build-actions/operates/Lienol/diy
2025-05-17T10:15:40.0109251Z   BUILD_FILES: /home/runner/work/build-actions/build-actions/operates/Lienol/files
2025-05-17T10:15:40.0109753Z   BUILD_PATCHES: /home/runner/work/build-actions/build-actions/operates/Lienol/patches
2025-05-17T10:15:40.0110285Z   BUILD_PARTSH: /home/runner/work/build-actions/build-actions/operates/Lienol/diy-part.sh
2025-05-17T10:15:40.0110817Z   BUILD_SETTINGS: /home/runner/work/build-actions/build-actions/operates/Lienol/settings.ini
2025-05-17T10:15:40.0111347Z   MYCONFIG_FILE: /home/runner/work/build-actions/build-actions/operates/Lienol/seed/x86_64
2025-05-17T10:15:40.0111720Z   DIY_WORK: Lienol2410
2025-05-17T10:15:40.0111925Z   DIY_PT1_SH: /tmp/common/Lienol/diy-part.sh
2025-05-17T10:15:40.0112184Z   DIY_PT2_SH: /tmp/common/Lienol/diy2-part.sh
2025-05-17T10:15:40.0112418Z   COMMON_SH: /tmp/common/common.sh
2025-05-17T10:15:40.0112654Z   UPGRADE_SH: /tmp/common/upgrade.sh
2025-05-17T10:15:40.0112887Z   CONFIG_TXT: /tmp/common/config.txt
2025-05-17T10:15:40.0113102Z   OpenClash_branch: 1
2025-05-17T10:15:40.0113293Z   Mandatory_theme: argon
2025-05-17T10:15:40.0113486Z   Default_theme: argon
2025-05-17T10:15:40.0113696Z   REPO_URL: https://github.com/Lienol/openwrt
2025-05-17T10:15:40.0113943Z   SOURCE: Lienol
2025-05-17T10:15:40.0114107Z   SOURCE_OWNER: Lienol
2025-05-17T10:15:40.0114295Z   DISTRIB_SOURCECODE: lienol
2025-05-17T10:15:40.0114494Z   LUCI_EDITION: 24.10
2025-05-17T10:15:40.0114941Z   GENE_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/bin/config_generate
2025-05-17T10:15:40.0115588Z   FILES_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/etc/shadow
2025-05-17T10:15:40.0116208Z   DELETE: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/etc/deletefile
2025-05-17T10:15:40.0116834Z   DEFAULT_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/auto-scripts/files/99-first-run
2025-05-17T10:15:40.0117574Z   KEEPD_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/lib/upgrade/keep.d/base-files-essential
2025-05-17T10:15:40.0118105Z   CLEAR_PATH: /tmp/Clear
2025-05-17T10:15:40.0118297Z   UPGRADE_DATE: 1747476898
2025-05-17T10:15:40.0118775Z   GUJIAN_DATE: 05.17
2025-05-17T10:15:40.0119086Z   LICENSES_DOC: /home/runner/work/build-actions/build-actions/openwrt/LICENSES/doc
2025-05-17T10:15:40.0119436Z   CON_TENTCOM: Lienol/openwrt
2025-05-17T10:15:40.0119776Z   RAW_WEB: https://raw.githubusercontent.com/Lienol/openwrt/24.10/feeds.conf.default
2025-05-17T10:15:40.0120124Z   SHANG_CHUAN: true
2025-05-17T10:15:40.0120297Z ##[endgroup]
2025-05-17T10:15:40.0139166Z ##[group]Run cd $HOME_PATH
2025-05-17T10:15:40.0139406Z [36;1mcd $HOME_PATH[0m
2025-05-17T10:15:40.0139607Z [36;1mbash $COMMON_SH Diy_menu[0m
2025-05-17T10:15:40.0184088Z shell: /usr/bin/bash --noprofile --norc -e -o pipefail {0}
2025-05-17T10:15:40.0184554Z env:
2025-05-17T10:15:40.0184743Z   GIT_REPOSITORY: AlieZ777/build-actions
2025-05-17T10:15:40.0185135Z   REPO_TOKEN: ***
[... 44 lines cut ...]
2025-05-17T10:15:40.0196621Z   DISTRIB_SOURCECODE: lienol
2025-05-17T10:15:40.0196815Z   LUCI_EDITION: 24.10
2025-05-17T10:15:40.0197228Z   GENE_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/bin/config_generate
2025-05-17T10:15:40.0197872Z   FILES_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/etc/shadow
2025-05-17T10:15:40.0198779Z   DELETE: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/etc/deletefile
2025-05-17T10:15:40.0199417Z   DEFAULT_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/auto-scripts/files/99-first-run
2025-05-17T10:15:40.0200148Z   KEEPD_PATH: /home/runner/work/build-actions/build-actions/openwrt/package/base-files/files/lib/upgrade/keep.d/base-files-essential
2025-05-17T10:15:40.0200679Z   CLEAR_PATH: /tmp/Clear
2025-05-17T10:15:40.0200873Z   UPGRADE_DATE: 1747476898
2025-05-17T10:15:40.0201183Z   GUJIAN_DATE: 05.17
2025-05-17T10:15:40.0201487Z   LICENSES_DOC: /home/runner/work/build-actions/build-actions/openwrt/LICENSES/doc
2025-05-17T10:15:40.0201841Z   CON_TENTCOM: Lienol/openwrt
2025-05-17T10:15:40.0202185Z   RAW_WEB: https://raw.githubusercontent.com/Lienol/openwrt/24.10/feeds.conf.default
2025-05-17T10:15:40.0202538Z   SHANG_CHUAN: true
2025-05-17T10:15:40.0202702Z ##[endgroup]
2025-05-17T10:15:40.0312783Z 
2025-05-17T10:15:40.0313397Z [0;33m正在执行：下载和整理应用,请耐心等候...[0m
2025-05-17T10:15:40.9380952Z auto-scripts文件下载完成
2025-05-17T10:15:40.9574340Z Cloning into '/tmp/tmp.nld2fjFpUt'...
2025-05-17T10:15:45.8140727Z luci-theme-argon文件下载完成
2025-05-17T10:16:52.7363127Z packages_lang_golang文件下载完成
2025-05-17T10:16:52.9067834Z feeds_packages_lang_node-prebuilt文件下载完成
2025-05-17T10:16:53.9188734Z packr文件下载完成
2025-05-17T10:16:54.3026638Z 增加定时更新固件的插件下载完成
2025-05-17T10:16:58.1103458Z ##[error]Process completed with exit code 1.

281677160, 2025-05-18:
应该没问题了，昨天在修改脚本的一个小问题，找了半天，改来改去，这个问题不影响编译，但是始终还是个问题，找了好久才找出来，昨天晚上2点多已经修复好了

drw2005, 2025-05-19:
最近两次编译，到“luci-theme-argon文件下载完成”处不进行下去了，直到回车一下才自动下去。。。

drw2005, 2025-05-19:
另外，luci-app-poweroffdevice怎么不见了？

281677160, 2025-05-19:
> 另外，luci-app-poweroffdevice怎么不见了？

有了

281677160, 2025-05-19:
> 最近两次编译，到“luci-theme-argon文件下载完成”处不进行下去了，直到回车一下才自动下去。。。

那是你网络卡了，修复了，以前是闭屏信息的，现在显示信息了，运行到那里都显示了

closed 2025-05-19 by 281677160