281677160/build-actions #217: 在线编译提示-缺少对比版本号文件

authon, 2025-05-20:
缺少对比版本号文件
https://github.com/authon/build-actions/actions/runs/15121157153/job/42503628224#:~:text=31-,%E7%BC%BA%E5%B0%91%E5%AF%B9%E6%AF%94%E7%89%88%E6%9C%AC%E5%8F%B7%E6%96%87%E4%BB%B6,-32

281677160, 2025-05-20:
重新编译看看

authon, 2025-05-21:
还是这样，一个多月了

281677160, 2025-05-21:
> 还是这样，一个多月了

你的密匙好像勾选的时候勾少了东西，只能下载到文件，不能下载到仓库

datout, 2025-05-21:
我也是 缺少对比版本号文件

281677160, 2025-05-21:
> 我也是 缺少对比版本号文件，创建密钥时候 全勾选了

你的不了解，上面那个仁兄肯定是密匙有问题，他的是下载不了自己的仓库，没办法同步更新，是密匙的问题吧，你们重新拉取最新的仓库吧

datout, 2025-05-21:
> > 我也是 缺少对比版本号文件，创建密钥时候 全勾选了
> 
> 你的不了解，上面那个仁兄肯定是密匙有问题，他的是下载不了自己的仓库，没办法同步更新，是密匙的问题吧，你们重新拉取最新的仓库吧

没事 重新fork了。没问题了

281677160, 2025-05-21:
这次更新，重新搞了很多东西，理顺了很多代码，在线更新那个插件也重新写过了，这次或者也是最后一次更新了，我自己好久没用openwrt了，就是手机去年年中更新了鸿蒙5.0后，没有可以用的翻墙软件，才重新使用openwrt的，以后有了可用的软件后，我也再也不需要使用openwrt了，翻墙的需求我也很少，如果你们最近使用当中发现什么问题，就提出来，能改就改好

281677160, 2025-05-21:
还有就是大雕的源码，昨天还是前天更新了6.6.91后，就编译不成功了，要等他修复

datout, 2025-05-21:
> 还有就是大雕的源码，昨天还是前天更新了6.6.91后，就编译不成功了，要等他修复

等大调修复好 测试下看看

authon, 2025-05-21:
> > 还是这样，一个多月了
> 
> 你的密匙好像勾选的时候勾少了东西，只能下载到文件，不能下载到仓库

好的，我再搞搞，之前编译是没问题的，好像是从上个月初开始这样的

hbzmt, 2025-05-22:
> 这次更新，重新搞了很多东西，理顺了很多代码，在线更新那个插件也重新写过了，这次或者也是最后一次更新了，我自己好久没用openwrt了，就是手机去年年中更新了鸿蒙5.0后，没有可以用的翻墙软件，才重新使用openwrt的，以后有了可用的软件后，我也再也不需要使用openwrt了，翻墙的需求我也很少，如果你们最近使用当中发现什么问题，就提出来，能改就改好

大佬，后续你不更新了，如果像immortalwrt更新大的版本号之后，我们这代码如何修改啊，想一直用最新的。

281677160, 2025-05-22:
> > 这次更新，重新搞了很多东西，理顺了很多代码，在线更新那个插件也重新写过了，这次或者也是最后一次更新了，我自己好久没用openwrt了，就是手机去年年中更新了鸿蒙5.0后，没有可以用的翻墙软件，才重新使用openwrt的，以后有了可用的软件后，我也再也不需要使用openwrt了，翻墙的需求我也很少，如果你们最近使用当中发现什么问题，就提出来，能改就改好
> 
> 大佬，后续你不更新了，如果像immortalwrt更新大的版本号之后，我们这代码如何修改啊，想一直用最新的。

更新版本不就是一直同步源码？还想怎么更新？如果有新的分支，你自己写上新分支就可以编译了啊，就跟增加机型文件一样意思