281677160/build-actions #222: cmcc rax3000m emmc编译固件后网络里没有无线选项

rogueme, 2025-05-29:
固件选择的是2410
大佬这个问题怎么解决呢，搜了下也没有wifi信号

281677160, 2025-05-30:
我这里没有对2410的源码有任何改动啊，23跟21有点改动，如果是2410有问题，你只能去源码作者那里问了

281677160, 2025-06-01:
> 找到问题了，是autosamba问题，我取消相关配置和软件包就有无线功能了

我都告诉你了，我这里没对2410源码做任何修改，你这样应该直接找源码的作者反映一下的，反正你自己都在用人家源码，你提出问题，人家修复，你以后用起来不是更好吗？如果是我这里有啥更改导致又不一样，你提出问题，我肯定去找原因

281677160, 2025-06-01:
> > > 找到问题了，是autosamba问题，我取消相关配置和软件包就有无线功能了
> > 
> > 
> > 我都告诉你了，我这里没对2410源码做任何修改，你这样应该直接找源码的作者反映一下的，反正你自己都在用人家源码，你提出问题，人家修复，你以后用起来不是更好吗？如果是我这里有啥更改导致又不一样，你提出问题，我肯定去找原因
> 
> https://github.com/281677160/build-actions/blob/main/build/Mt798x/diy-part.sh 这里有去除网络共享的选项就是关于autosamba，默认是不修改，我不知道实际有没有去除，我改成1去除了！ https://github.com/281677160/build-actions/blob/main/build/Mt798x/seed/cmcc-rax3000m-emmc 这里CONFIG_PACKAGE_autosamba=y，CONFIG_PACKAGE_luci-app-samba4=y，CONFIG_PACKAGE_luci-i18n-samba4-zh-cn=y，CONFIG_SAMBA4_SERVER_AVAHI=y CONFIG_SAMBA4_SERVER_NETBIOS=y CONFIG_SAMBA4_SERVER_VFS=y CONFIG_SAMBA4_SERVER_WSDD2=y均是默认安装，我全部去掉编译后就有无线功能了！

默认的配置文件我都是直接源码作者那里复制过来的，我没有这些设备，对这些设置不熟悉，所以也没乱改什么，23跟21的代码因为编译的时候跟24的配置文件可以通用，倒是改了写代码来适应