281677160/build-actions #224: 大佬 编译出错了

shengonglin, 2025-05-30:
[image]

大佬，新编译出现这个错误提示，然后编译了一会就报错了。你空了能看看是咋回事么，感谢哈~~

shengonglin, 2025-05-30:
关键是我不需要这个ssr-plus的插件。我也没选。我根本就不需要他。。。。

281677160, 2025-05-30:
> 关键是我不需要这个ssr-plus的插件。我也没选。我根本就不需要他。。。。

rust替换成旧版本了，你等几分钟再编译看看，正在弄代码，几分钟后有新版本更新

shengonglin, 2025-05-30:
> > 关键是我不需要这个ssr-plus的插件。我也没选。我根本就不需要他。。。。
> 
> rust替换成旧版本了，你等几分钟再编译看看，正在弄代码，几分钟后有新版本更新

好的。谢谢大佬，每一次问题你都能解决。真的好崇拜你！！

shengonglin, 2025-05-30:
大佬。抱歉，我又开始编译了。但是又发生了个问题。在插件列表里面。没有openclash了。所以是替换了rust的旧版本以后，插件就没有openclash了是么。。。

[image]

没替换之前，ssh选择插件列表还能看到的。现在看不到了，没得选了。。。

281677160, 2025-05-30:
openclash在diy-part.sh控制

closed 2025-05-30 by shengonglin